negativo17/nvidia-driver #85: 435.21 driver doesn't work with external monitor on Optimus laptop

pjuhasz, 2019-09-16:
I use Fedora 30 (kernel 5.2.14-200.fc30.x86_64) on a Thinkpad P50, which has an integrated Intel GPU and a separate Nvidia Quadro GPU. I use an external monitor connected to the HDMI output in addition to the laptop's own display. Until yesterday (with the 430.something driver version) the setup worked well: the laptop display was managed by the integrated GPU while the external monitor was driven by the Nvidia. nvidia-settings detected the laptop display as "PRIME display". 

The update to driver version 435.21 broke this. The external monitor did not come out from power save, xrandr didn't even detect it, and nvidia-settings wasn't able to start properly.

I was able to find a workaround here: https://rpmfusion.org/Howto/Optimus - where it says to add `Option "PrimaryGPU" "yes" ` to the `nvidia.conf` file located in `/etc/X11/xorg.conf.d`.
With this, the external monitor works again.

I think the reason of the breakage is hinted at in this driver version's official documentation: http://download.nvidia.com/XFree86/Linux-x86_64/435.21/README/primerenderoffload.html

"As of this writing, these commits are only in the master branch of the X.Org X server, and not yet in any official X.Org X server release."

scaronni, 2019-09-18:
Can you try to to comment the following line:

```
# cat /usr/lib/modprobe.d/nvidia.conf | grep NV
options nvidia NVreg_DynamicPowerManagement=0x02
```

Then running `depmod -a` and rebooting? This makes sure the Nvidia card does not power off completely, it can just reach the lowest power state.

If this solves it, I will make a note about the mux vs muxless laptops

pjuhasz, 2019-09-18:
Commenting out that line had no effect.

I undid my previous PrimaryGPU modification and rebooted to restore the known bad state, then followed your steps. The external monitor remained off, xrandr didn't detect any of the external outputs, so, no effect.

scaronni, 2019-09-19:
`"As of this writing, these commits are only in the master branch of the X.Org X server, and not yet in any official X.Org X server release."`

These have been pushed to the current X.org server package in Fedora 30+. Did you update recently?

```
$ rpm -q --changelog xorg-x11-server-Xorg | head -2
* Thu Aug 29 2019 Olivier Fourdan <ofourdan@redhat.com> 1.20.5-7
- Pick latest fixes from xserver stable branch upstream (rhbz#1729925)
```

If you set "PrimaryGPU", then the card is, well the primary one. Kind of defeats the point of the whole PRIME Offload thing. Can you try the snippet at the end of this page http://download.nvidia.com/XFree86/Linux-x86_64/435.21/README/primerenderoffload.html to force the GPU screens to be created?

If automatic configuration does not work, it may be necessary to explicitly configure the iGPU and dGPU devices in xorg.conf:
```
    Section "ServerLayout"
      Identifier "layout"
      Screen 0 "iGPU"
      Option "AllowNVIDIAGPUScreens"
    EndSection

    Section "Device"
      Identifier "iGPU"
      Driver "modesetting"
    EndSection

    Section "Screen"
      Identifier "iGPU"
      Device "iGPU"
    EndSection

    Section "Device"
      Identifier "dGPU"
      Driver "nvidia"
    EndSection
```

TauAkiou, 2019-10-11:
I have this same issue, and I theorize that it's because our video outputs are wired to the NVIDIA card, which is theoretically not being used in this configuration unless activated. 

Like Optimus under Windows, all of the standard rendering is being handled by the igfx and the NVIDIA card remains in standby unless explicitly called. Under Windows, I assume they have a method in place to handle this, but under Linux it appears that NVIDIA didn't anticipate this.

Naturally, these outputs will work when the GPU is set to Primary as the GPU is both active & handling video output from those display outputs.

This is a hunch, but try using:

```
xrandr --listproviders

Provider 0: id: 0x271 cap: 0x1, Source Output crtcs: 4 outputs: 3 associated providers: 1 name:NVIDIA-0
Provider 1: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 1 name:modesetting

$ xrandr --setprovideroutoutsource <nvidia provider id> <intel provider id>```

scaronni, 2019-10-30:
> Set Option "PrimaryGPU" "Yes"in your xorg configuration like you've been doing, which reverts to the original functionality and makes the ports work properly.

Yeah, in this case the whole power off thing will not work. I guess there is no other chance than select one or the other.

closed 2020-02-15 by scaronni