negativo17/nvidia-driver #86: systemd-rpm-macros dependency on RHEL 8

tbaederr, 2019-09-25:
Hi,

in the last release, the nvidia-driver package has switched from depending on systemd to depending on systemd-rpm-macros at buildtime on RHEL 8. That package does not seem to exist on RHEL 8 however, should it instead keep depending on systemd and only depend on systemd-rpm-macros on Fedora? Does the package get built anywhere on RHEL 8 with this dependency?

scaronni, 2019-09-26:
Hello, I haven't build any package yet for CentOS/RHEL 8 except a few exceptions. Now that CentOS 8 is available and has reached the basic Mock configuration set, I will do it for all packages:

https://koji.fedoraproject.org/koji/buildinfo?buildID=1390472

tbaederr, 2019-10-01:
Is this fixed by https://github.com/negativo17/nvidia-driver/commit/1ed6ba25a70c074dc9a88b66edb3a16c54b657a1 so this can be closed, or are we waiting for a build that succeds?

scaronni, 2019-10-01:
I will close it once the builds are online (max 1 hour from now). If you could test them, it would be great.

scaronni, 2019-10-01:
It also required this:

https://github.com/negativo17/nvidia-kmod-common/commit/d44bdae594b8c01770d3b06854e6c82ea3f7406c

closed 2019-10-01 by scaronni