negativo17/nvidia-driver #87: Fedora 31 - no 32bit libs

silviucc, 2019-10-30:
The repository for the nvidia driver currently does not have the 32bit libs.

```
$ sudo dnf install nvidia-driver-libs.i686

Last metadata expiration check: 0:18:31 ago on Mi 30 oct 2019 17:38:04 +0200.
No match for argument: nvidia-driver-libs.i686
Error: Unable to find a match: nvidia-driver-libs.i686

$ sudo dnf search nvidia-driver-libs
Last metadata expiration check: 0:18:52 ago on Mi 30 oct 2019 17:38:04 +0200.
====================================== Name Exactly Matched: nvidia-driver-libs ======================================
nvidia-driver-libs.x86_64 : Libraries for nvidia-driver
nvidia-driver-libs.x86_64 : Libraries for nvidia-driver
```

scaronni, 2019-11-01:
Package pushed. I have a new baby coming this week, had multiple hardware failures and a tough time at work :)

Fedora 31 is missing still a couple of packages (HandBrake for example), but should be mostly in sync now along with Steam.

closed 2019-11-01 by scaronni

adatum, 2019-11-01:
Congratulations, and I hope the rest of your week will be much better!

Thanks for pushing the fixes so quickly.