negativo17/nvidia-driver #88: Driver deps non-existent

jeffclay, 2019-11-03:
Driver depends on libva-vdpau-driver which, as far as I can tell, doesn't exist for el8. This was encountered when trying to install on RHEL 8.

scaronni, 2019-11-03:
Still building all packages for RHEL/CentOS 8. Many, many dependencies missing. They just now removed the forbidden download (was getting a 403) on the packages listed in "known missing" at https://koji.mbox.centos.org/:

https://lwn.net/Articles/796202/

Still, though, they are still not available for consumption in EPEL, so many builds are blocked. I'm keeping an eye on them and trigger builds accordingly.

closed 2019-11-03 by scaronni

scaronni, 2019-11-03:
`libva-vdpau-driver` is now pending in the updates: https://bodhi.fedoraproject.org/updates/FEDORA-EPEL-2019-27380b2637

jeffclay, 2019-11-03:
libva-vdpau-driver was added after I submitted a bug report at https://bugzilla.redhat.com/show_bug.cgi?id=1768148 Do you think we need to submit bugs for all missing deps?