negativo17/nvidia-driver #89: nvidia-driver blocking upgrade to Fedora 31

adatum, 2019-11-03:
The fix in https://github.com/negativo17/nvidia-driver/issues/87 reduced the problems, but I am still seeing issues after `$ sudo dnf system-upgrade download --refresh --releasever=31`:

```
Error: 
 Problem: problem with installed package nvidia-driver-3:435.21-2.fc30.x86_64
  - package nvidia-driver-3:435.21-1.fc31.x86_64 conflicts with xorg-x11-drv-nvidia-304xx provided by rpmfusion-nonfree-obsolete-packages-31-1.fc31.noarch
  - package nvidia-driver-3:435.21-2.fc31.x86_64 conflicts with xorg-x11-drv-nvidia-304xx provided by rpmfusion-nonfree-obsolete-packages-31-1.fc31.noarch
  - problem with installed package gstreamer-plugins-bad-nonfree-0.10.23-9.fc30.x86_64
  - nvidia-driver-3:435.21-2.fc30.x86_64 does not belong to a distupgrade repository
  - gstreamer-plugins-bad-nonfree-0.10.23-9.fc30.x86_64 does not belong to a distupgrade repository
(try to add '--allowerasing' to command line to replace conflicting packages or '--skip-broken' to skip uninstallable packages)

```

Is this possibly related to the negativo repo or something else? I don't have `xorg-x11-drv-nvidia` installed on my system. I have done two Fedora release upgrades in the past on the same system without issues.

I also checked what would happen with `--allowerasing` and at a glance it seemed `nvidia-driver` would not get installed, neither would `xorg-x11-drv-nvidia` but all kinds of other nvidia and CUDA packages would be, as expected, from the negativo repo, but at least one, `nvidia-settings` would be coming from rpmfusion. That mess worried me and aborted it.

leigh123linux, 2019-11-03:
Why report this?, the only issue I can see is you.
Don't mix conflicting repo's!

leigh123linux, 2019-11-03:
Easy solution is to remove gstreamer-plugins-bad-nonfree package to avoid installing it's replacement (rpmfusion-nonfree-obsolete-packages).

https://pkgs.rpmfusion.org/cgit/nonfree/rpmfusion-nonfree-obsolete-packages.git/commit/?h=f31

scaronni, 2019-11-03:
@adatum the repositories here are not compaible with RPMFusion. If you want to use RPMFusion, remove all repositories hosted here.

closed 2019-11-03 by scaronni

adatum, 2019-11-04:
Thanks, removing `gstreamer-plugins-bad-nonfree` got rid of all problems blocking the upgrade, including the nvidia ones. Upgrading to Fedora 31 went smoothly after that.

As for the other comments, I'm a bit confused. Are you suggesting that the RPMFusion repos cannot be used at all if negativo repo is used for nvidia drivers? To be clear, I am not using nvidia packages from RPMFusion, but I do make use of other packages from that repo.

scaronni, 2019-11-04:
If you just use the nvidia repository from here and RPMFusion it should be fine.