negativo17/nvidia-driver #90: display issues after installing on RHEL 8

jeffclay, 2019-11-03:
I have installed nvidia driver on RHEL 8 after installing libva-vdpau-driver from https://koji.fedoraproject.org/koji/buildinfo?buildID=1407202 .

After installation I did not see the expected kernel building processes when running 'top'.
Prior to installation of this driver, nouveau showed display in proper resolution on 3 monitors plugged in to the same gpu. Which left 3 monitors on the 2nd gpu blank.
After installation of this driver, display was only shown on 1 monitor at what appeared to be800x600 resolution.

Driver was installed on RHEL 8 by running: 'yum install nvidia-driver' 
This pulled the following packages and versions:
nvidia-driver                       x86_64          3:435.21-2.el8
kmod-nvidia                       x86_64          3:435.21-1.el8
nvidia-driver-libs                 x86_64          3:435.21-2.el8
nvidia-kmod-common        noarch           3:435.21-3.el8
egl-wayland                       x86_64           1.1.2-1.el8
mesa-vulkan-drivers          x86_64           18.3.1-5.el8_0
vulkan_loader                    x86_64           1.1.101.0-0.el8_0

scaronni, 2019-11-04:
> After installation I did not see the expected kernel building processes when running 'top'.
> kmod-nvidia x86_64 3:435.21-1.el8

This is because the `kmod-nvidia` package contains just binary modules.

I'm pushing updated packages in an hour or so, please also install `nvidia-settings` and check if the modules are properly loaded and Nouveau is disabled. Are the 2 GPUs all Nvidia cards?

Would be curious to know what you need to do with 6 monitors. Crypto trading? :)

jeffclay, 2019-11-05:
Thanks for pushing the packages. All seems to be working well. There is one small quirk though. When viewing the displays to change the alignment, the monitor graphics are very small. https://imgur.com/a/QniUJMy 

Now, I'm just looking forward to the Negativo17 epel repos to catch up to the F31 repo packages.

Also, 6 monitors are much better than 2; so much more content at once.

scaronni, 2019-11-05:
The Nvidia packages should be on par with the Fedora ones, I will build 440.31 in the next days.

Multimedia packages will still take a while as EPEL has some dependencies missing due to some missing devel packages missing in RHEL 8.

Btw, I don't think I have enough content for 6 monitors. :D

closed 2019-11-05 by scaronni