negativo17/nvidia-driver #93: Make nvidia-driver-cuda depend on nvidia-driver-NVML?

tbaederr, 2019-12-04:
Hi,

I'm currently looking into a problem that appears when using a negativo-style repository in conjunction with nvidia's cuda repository, on RHEL 8 (it is not possible to do this with negativo+cuda right now).

The `nvidia-driver-cuda` package has an automatic dependency on `libnvidia-ml.so.1()(64bit)`, which is usually shipped by `nvidia-driver-NVML`. This dependency is important since `nvidia-smi` will not work without the `libnvidia-ml.so`:

```
NVIDIA-SMI couldn't find libnvidia-ml.so library in your system. Please make sure that the NVIDIA Display Driver is properly installed and present in your system.
Please also try adding directory that contains libnvidia-ml.so to your system PATH.
```
However, there is a package called `cuda-nvml-dev-10-1` (and `-10-2`) in the cuda repository [here](http://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64/cuda-nvml-dev-10-1-10.1.243-1.x86_64.rpm) which also ships `/usr/local/cuda-10.2/targets/x86_64-linux/lib/stubs/libnvidia-ml.so` and thereore satisfies this dependency, causing an installation with broken `nvidia-smi`.

I wanted to ask upstream here whether this analysis is correct and whether an explicit dependency from `nvidia-driver-cuda` to `nvidia-driver-NVML` is the proper solution to solve such issues.

scaronni, 2019-12-09:
Hi,

so the binary should depend on the versioned library and not the versioned one. I see this on the binary:
```
$ strings /usr/bin/nvidia-smi | grep libnvidia-ml.so
libnvidia-ml.so
/usr/lib64/libnvidia-ml.so.1
/usr/lib/libnvidia-ml.so.1
/usr/lib32/libnvidia-ml.so.1
/usr/lib64/nvidia/libnvidia-ml.so.1
/usr/lib/nvidia/libnvidia-ml.so.1
```
Does this mean it's ok already and those are hardcoded attempted locations?

Regarding the `cuda-nvml-dev*` packages, these should really not provide the stub libraries. The point of the automatic library dependency generator in RPM is exactly that, that the packages automatically list requirements and provides for dependency resolution. If the stub library is used only for compilation and not for runtime, it should be explicitly filtered out in the SPEC file:

https://docs.fedoraproject.org/en-US/packaging-guidelines/AutoProvidesAndRequiresFiltering/

tbaederr, 2020-05-05:
the `libnvidia-ml.so` in `cuda-nvml-dev` is a stub and thus does not provide the functionality and should be filtered out. This needs to be fixed on the nvidia side. Thanks and sorry this took so long.

closed 2020-05-05 by tbaederr