negativo17/nvidia-driver #94: Fedora 31: Optimus Laptop has no access to nvidia driver

exoticDFT, 2019-12-08:
I have spent the last week+ trying to get the Prime functionality working with my Optimus laptop with a RTX 2080 working in Fedora 31, but have been unable to successfully use the NVIDIA card. No matter what combination of X configurations discussed in the previously posted issues, it will not access the card. It almost seems like the x-server is not really loading correctly or that it is ignoring all configurations.

To be more specific, I have tried:
- The default configuration
- Explicitly configuring xorg.conf with [iGPU config](http://download.nvidia.com/XFree86/Linux-x86_64/435.17/README/primerenderoffload.html)
- Modifying 10-nvidia.conf with [PrimaryGPU option](https://rpmfusion.org/Howto/Optimus)
- Creating an xorg.conf file from http://us.download.nvidia.com/XFree86/Linux-x86/375.26/README/randr14.html

Initial steps before messing with the X configuration:
1. Fresh install of Fedora 31
2. dnf update
3. Reboot
4. Add negativo17 repo
5. Installed nvidia-driver, nvidia-driver-libs.i686, dkms-nvidia, nvidia-settings, cuda-devel, nvidia-driver-cuda
6. Reboot

Then output from some commands that suggests nothing is working:

`$ __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia nvidia-smi`
> NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.

`$ DRI_PRIME=1 glxinfo | grep "OpenGL renderer"`
> OpenGL renderer string: Mesa DRI Intel(R) UHD Graphics 630 (Coffeelake 3x8 GT2)

`$ __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia glxinfo | grep "OpenGL renderer"`
> X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  150 (GLX)
  Minor opcode of failed request:  24 (X_GLXCreateNewContext)
  Value in failed request:  0x0
  Serial number of failed request:  39
  Current serial number in output stream:  40

`$ lsmod | grep nouveau`
Returns nothing

`$ lsmod | grep nvidia`
> i2c_nvidia_gpu         16384  0

`$ lspci | grep VGA`
> 00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 630 (Mobile)
01:00.0 VGA compatible controller: NVIDIA Corporation TU104M [GeForce RTX 2080 Mobile] (rev a1)

`xrandr --listproviders`
> Providers: number : 0

No matter which configuration I choose, I get the same results as above. Any help would be greatly appreciated.

scaronni, 2020-01-29:
> `$ lsmod | grep nvidia`
> 
> > i2c_nvidia_gpu         16384  0

Sorry missed completely this issue. Here simply the kernel module is not loaded (so no driver working):
```
# lsmod | grep ^nvidia
nvidia_drm             49152  0
nvidia_modeset       1118208  1 nvidia_drm
nvidia_uvm           1081344  0
nvidia              20439040  44 nvidia_uvm,nvidia_modeset
```

To be precise, we manage a fleet of Fedora Optimus muxless laptops (i.e. all outputs are anyway attached to the Intel GPU) and we have no issue at all with the default configuration (just the package installation). This is what we have on laptops:
```
# rpm -qa \*nvidia\* | sort
akmod-nvidia-440.44-1.fc31.x86_64
kmod-nvidia-5.4.13-201.fc31.x86_64-440.44-1.fc31.x86_64
nvidia-driver-440.44-1.fc31.x86_64
nvidia-driver-libs-440.44-1.fc31.i686
nvidia-driver-libs-440.44-1.fc31.x86_64
nvidia-kmod-common-440.44-1.fc31.noarch
```

scaronni, 2020-01-29:
> If you're running kernel 5.4.x, it won't work. You'll have to wait for NVidia to release a new driver compatible with kernel 5.4+. They released a Vulkan beta driver which appears to be compatible:
> https://developer.nvidia.com/vulkan-driver
> There's also a non-official patch that should fix it (look for Isaak Aleksandrov post):
> https://devtalk.nvidia.com/default/topic/1068045/linux/5-4-kernel-breaks-prime-synchronization-/
> It would be nice if the negativo17 repo updated the driver with that patch (at least temporarily). Fedora moved to the 5.4 series a while ago which breaks PRIME for quite a few.

At work we've not enabled PRIME synchronization as most of the people (even though they have the choice) just run Gnome on Wayland. You can still offload applications to the Nvidia card like this:
```
alias nvidia-run='XDG_SESSION_TYPE=X11 __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia "$@"'
```
This makes sure that offloaded applications run on Xwayland, which is absolutely fine.

Regarding the Vulkan drivers with the fix that you pointed out:
[image]

That would be perfect and I would do it if only Nvidia would release source code for the various open source bits also for beta/vulkan drivers, unfortunately they don't and then I would need to rework completely the packaging even if temporarily.

I will look at the patch.

kmare, 2020-01-29:
@scaronni First of all, thank you for your work on the repo. It's truly appreciated! 
The truth is that this is clearly a problem that the NVidia driver itself should fix. I was expecting them to release a fix much sooner, but no luck till now and I'm simply staying with the kernel 5.3 series. Of course, if you included that patch, it would be great!
Thank you for the additional information btw, it's definitely useful.

leigh123linux, 2020-01-29:
> If you're running kernel 5.4.x, it won't work. You'll have to wait for NVidia to release a new driver compatible with kernel 5.4+. They released a Vulkan beta driver which appears to be compatible: [...]

The patch is garbage, it would break compatibility with older kernels.

kmare, 2020-01-29:
@leigh123linux you're right, I just took a quick look at it, and it doesn't perform any kernel versions checks. Either someone implements that, or we just need to wait for a proper NVidia fix :/

leigh123linux, 2020-02-02:
@scaronni You dropped the nvidia-prime.patch, did it have flaws?
I have backported for the vulkan developer driver if your interested

https://pkgs.rpmfusion.org/cgit/nonfree/nvidia-kmod.git/tree/nvidia_backport.patch

It also adds kernel-5.5 support, dmesg has some non fatal kernel messages for 5.5

kmare, 2020-02-03:
Well, a new driver with proper 5.4+ support was just released:
https://www.nvidia.com/download/driverResults.aspx/156772/en-us
Hopefully we'll see it in the repo soon :)

scaronni, 2020-02-13:
> @scaronni You dropped the nvidia-prime.patch, did it have flaws?
> I have backported for the vulkan developer driver if your interested [...]

Sorry, just saw the message now. Thanks, in the end I just waited for the new driver which came out immediately after.

closed 2020-02-13 by scaronni