negativo17/nvidia-driver #100: Installation of CuDNN

sbrl, 2020-06-16:
Hello,

I'm using Fedora 31 with the negativo17 nvidia drivers. The installation process I used is documented here: https://forums.freeside.co.uk/t/installing-nvidia-drivers-on-fedora-30/193

The driver itself works fine, and I can use `nvidia-smi` to see the GPU status:

```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.82       Driver Version: 440.82       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 750 Ti  Off  | 00000000:01:00.0 Off |                  N/A |
| 31%   27C    P8     1W /  38W |    214MiB /  2001MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1267      G   /usr/libexec/Xorg                             73MiB |
|    0      1390      G   /usr/bin/gnome-shell                         134MiB |
|    0      1658      G   /usr/libexec/gnome-initial-setup               1MiB |
+-----------------------------------------------------------------------------+
```

However, if I try to run a Tensorflow.js program, it fails to find the following libraries:

```
libcudart.so.10.0
libcublas.so.10.0
libcufft.so.10.0
libcurand.so.10.0
libcusolver.so.10.0
libcusparse.so.10.0
libcudnn.so.7
```

(full Tensorflow.js error output: https://ybin.me/p/2089b700cc596ac2#pC7BEMbn5XvMy4X+CQkTbToW0VkEmriAx/vjZeiRQf0=)

With this in mind, my research suggests that installing "CuDNN" will fix the issue. However, I am aware that if I accidentally install CuDNN from a source other than the negativo17 drivers, this may potentially conflict with the negativo drivers.

**How can I install CuDNN from the negativo 3rd party repo?**

scaronni, 2020-06-17:
You're mixing things.

> However, if I try to run a Tensorflow.js program, it fails to find the following libraries:
> 
> ```
> libcudart.so.10.0
> libcublas.so.10.0
> libcufft.so.10.0
> libcurand.so.10.0
> libcusolver.so.10.0
> libcusparse.so.10.0

These are CUDA 10.0 libraries, I'm shipping CUDA 10.2:
```
/usr/lib64/libcublas.so.10.2.2.89
/usr/lib64/libcublasLt.so.10.2.2.89
/usr/lib64/libnvblas.so.10.2.2.89
/usr/lib64/libcudart.so.10.2
/usr/lib64/libcudart.so.10.2.89
/usr/lib64/libcufft.so.10.1.2.89
/usr/lib64/libcufftw.so.10.1.2.89
/usr/lib64/libcupti.so.10.2
/usr/lib64/libcupti.so.10.2.75
/usr/lib64/libcurand.so.10.1.2.89
/usr/lib64/libcusolver.so.10.3.0.89
/usr/lib64/libcusolverMg.so.10.3.0.89
/usr/lib64/libcusparse.so.10.3.1.89
/usr/lib64/libaccinj64.so.10.2
/usr/lib64/libaccinj64.so.10.2.89
/usr/lib64/libcudart.so.10.2
/usr/lib64/libcudart.so.10.2.89
/usr/lib64/libcuinj64.so.10.2
/usr/lib64/libcuinj64.so.10.2.89
/usr/lib64/libnppc.so.10.2.1.89
/usr/lib64/libnppial.so.10.2.1.89
/usr/lib64/libnppicc.so.10.2.1.89
/usr/lib64/libnppicom.so.10.2.1.89
/usr/lib64/libnppidei.so.10.2.1.89
/usr/lib64/libnppif.so.10.2.1.89
/usr/lib64/libnppig.so.10.2.1.89
/usr/lib64/libnppim.so.10.2.1.89
/usr/lib64/libnppist.so.10.2.1.89
/usr/lib64/libnppisu.so.10.2.1.89
/usr/lib64/libnppitc.so.10.2.1.89
/usr/lib64/libnpps.so.10.2.1.89
/usr/lib64/libnvgraph.so.10.2.89
/usr/lib64/libnvjpeg.so.10.3.1.89
/usr/lib64/libnvrtc-builtins.so.10.2
/usr/lib64/libnvrtc-builtins.so.10.2.89
/usr/lib64/libnvrtc.so.10.2
/usr/lib64/libnvrtc.so.10.2.89
```

> libcudnn.so.7

For this just install `cuda-cudnn`:

```
$ rpm -qpl cuda-cudnn-7.6.5.32-1.fc32.x86_64.rpm 
/usr/lib64/libcudnn.so.7
/usr/lib64/libcudnn.so.7.6.5
/usr/share/licenses/cuda-cudnn
/usr/share/licenses/cuda-cudnn/NVIDIA_SLA_cuDNN_Support.txt
```

> (full Tensorflow.js error output: https://ybin.me/p/2089b700cc596ac2#pC7BEMbn5XvMy4X+CQkTbToW0VkEmriAx/vjZeiRQf0=)
>
> With this in mind, my research suggests that installing "CuDNN" will fix the [...]
```
yum -y install cuda-cudnn
```
But make sure you are using a TensorFlow compiled for CUDA 10.2.

closed 2020-06-17 by scaronni

sbrl, 2020-06-21:
Thanks, @scaronni! I'll try that.

I assume that will work with `dnf` too.

I'm having issues compiling Tensorflow too, but this isn't the right place to debug such issues and they are not limited to Fedora and this repo.