negativo17/nvidia-driver #104: Error after update to  455.28

pablomenino, 2020-10-13:
I update the nvidia package to 455.28 (dnf update) 

Can't boot into graphical mode, 

I try to remove and install again with dkms, try to boot older kernels. And nothing. 

I can't catch the error

I am on KDE, kernel 5.8.14-200

Any way to rollback to version 450.80.02

I try with dnf - - showduplicates, but only see the last nvidia package.

pablomenino, 2020-10-13:
after remove nvidia driver the system is working with Nouveau

h3o66, 2020-10-29:
Did you install the driver with dkms or akmod ?

pablomenino, 2020-10-29:
dkms

sudo dnf install nvidia-settings kernel-devel dkms-nvidia nvidia-driver-libs.i686 vulkan-loader.x86_64 vulkan-loader.i686

pablomenino, 2021-02-26:
Solved after migrated to RPM Fusion driver package

closed 2021-02-26 by pablomenino