negativo17/nvidia-driver #108: unsigned drivers since version 460+

graphicsMan, 2021-02-11:
Thanks for the handy repos for installing nvidia drivers.  In recent versions, I've had issues where the kernel is angry at the NVIDIA driver because it is proprietary and unsigned (Fedora 33).  These same versions also have an issue where every time I run a CUDA program there is a 9ish second lag before the initialization can occur.  After init, things seem to work reasonably.  Although I'm not 100% sure these two things are tied together, I can say that in the 455 driver, I had neither symptom.

In the journal logs I see something like this after I run nvidia-smi

Feb 10 16:56:18 my_machine kernel: NVRM: GPU 0000:09:00.0: rm_init_adapter failed, device minor number 0
Feb 10 16:56:18 my_machine kernel: NVRM: GPU 0000:09:00.0: RmInitAdapter failed! (0x26:0x65:1290)
Feb 10 16:56:13 my_machine kernel: NVRM: GPU 0000:09:00.0: rm_init_adapter failed, device minor number 0
Feb 10 16:56:13 my_machine kernel: NVRM: GPU 0000:09:00.0: RmInitAdapter failed! (0x26:0x65:1290)

Is it possible to sign these drivers prior to pushing version updates into the repo?  And/or do you know of any other possible causes of the init failure (which eventually seems to be bypassed for successful init, just slowly)?

scaronni, 2021-02-11:
Hello,

If I sign the drivers, you would need to manually enroll the keys in your TPM device. And the source RPMs could just not be rebuilt as they are, as they would not cointain the key. So no, you have to decide if you want to use the drivers and disable secure boot, or sign the drivers yourself through DKMS or akmods.

Still applies: https://docs.fedoraproject.org/en-US/Fedora/23/html/System_Administrators_Guide/sect-kernel-module-authentication.html#table-kernel-module-authentication-requirements-for-loading

The error you are facing has nothing to do with signing, if it would be a signing issue you would not be allowed to load the module at all.

graphicsMan, 2021-02-11:
I see.  Thank you for the explanation.

closed 2021-02-11 by graphicsMan