negativo17/nvidia-driver #116: Unable to use nvidia utilities after update to 495 driver

TurboTurtle, 2021-11-08:
I updated to the 495 driver, specifically:

```
nvidia-libXNVCtrl-495.44-1.fc34.x86_64
nvidia-driver-libs-495.44-1.fc34.x86_64
nvidia-driver-495.44-1.fc34.x86_64
nvidia-driver-cuda-libs-495.44-1.fc34.x86_64
nvidia-driver-NVML-495.44-1.fc34.x86_64
nvidia-persistenced-495.44-1.fc34.x86_64
nvidia-driver-cuda-495.44-1.fc34.x86_64
nvidia-settings-495.44-1.fc34.x86_64
```

After which I am unable to use any nvidia utilities, e.g.:

```
$ nvidia-settings

ERROR: Unable to find display on any available system

$ nvidia-smi
NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.
```

Further, no games will launch successfully. Most will end up with a black or white screen. Interestingly however, GNOME works perfectly fine and I can drive 3 monitors at 1920x1080 without issue via the card. 

Fedora 34
Kernel 5.14.16-201.fc34.x86_64
GTX 980

scaronni, 2021-11-08:
Works fine here on an Xorg session. Are you using a Wayland session?

scaronni, 2021-11-08:
I not, can you also paste the output of `dmesg | grep -i nvidia` just after rebooting? Thanks.

TurboTurtle, 2021-11-08:
On Xorg. Downgrading to the 470.82 driver restores normal functionality, I'll try upgrading again later today and report back.

combolek, 2021-11-08:
Nvidia 495.* drivers do not support GTX 980 anymore, so you have to use the 470.* series. I am in the same boat.

Would it be possible to make 470.* drivers available for Fedora 35?
Thanks

kmare, 2021-11-08:
The GTX 980 is supported by the 495 series. I have the GTX 960 and it works fine, both laptop and desktop.
See:
https://download.nvidia.com/XFree86/Linux-x86_64/495.44/README/supportedchips.html

combolek, 2021-11-08:
OK, this is confusing. I have a GTX 770 which also on this list and it is definitely not supported (gives a specific error message on startup saying exactly that).
So maybe 9xx is supported but 7xx is not? But this list is suspect.

In any case, it would use useful to have 470.* drivers available for older cards.
Thanks

combolek, 2021-11-08:
Hmm, sorry I misread. GTX 770 is on a different list (for 470). So yeah, it looks like 9xx is still supported in 495 but 7xx is not.

TurboTurtle, 2021-11-09:
After a reinstall of the 495 drivers I'm having different behavior now.

`nvidia-smi` and `nvidia-settings` work properly, however any DX11/12 game (via steam, thus Proton) will report an error of `DX11 feature level 10.0 is required to run the engine`.

If relevant:

```
$ nvidia-smi
Mon Nov  8 22:25:15 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 495.44       Driver Version: 495.44       CUDA Version: 11.5     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:0A:00.0  On |                  N/A |
|  0%   49C    P0    49W / 196W |    305MiB /  4040MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2669      G   /usr/libexec/Xorg                 111MiB |
|    0   N/A  N/A      2814      G   /usr/bin/gnome-shell              140MiB |
|    0   N/A  N/A      3779      G   ...AAAAAAAAA= --shared-files       48MiB |
+-----------------------------------------------------------------------------+

$ dmesg | grep -i nvidia
[    5.779971] input: HDA NVidia HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:03.1/0000:0a:00.1/sound/card1/input15
[    5.780045] input: HDA NVidia HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:03.1/0000:0a:00.1/sound/card1/input16
[    5.780124] input: HDA NVidia HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:03.1/0000:0a:00.1/sound/card1/input17
[    5.780179] input: HDA NVidia HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:03.1/0000:0a:00.1/sound/card1/input18
[    5.780226] input: HDA NVidia HDMI/DP,pcm=10 as /devices/pci0000:00/0000:00:03.1/0000:0a:00.1/sound/card1/input19
[    5.780363] input: HDA NVidia HDMI/DP,pcm=11 as /devices/pci0000:00/0000:00:03.1/0000:0a:00.1/sound/card1/input20
[    5.780409] input: HDA NVidia HDMI/DP,pcm=12 as /devices/pci0000:00/0000:00:03.1/0000:0a:00.1/sound/card1/input21
[    6.622785] nvidia: loading out-of-tree module taints kernel.
[    6.622798] nvidia: module license 'NVIDIA' taints kernel.
[    6.634169] nvidia: module verification failed: signature and/or required key missing - tainting kernel
[    6.651254] nvidia-nvlink: Nvlink Core is being initialized, major device number 510
[    6.652486] nvidia 0000:0a:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[    6.855143] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  495.44  Fri Oct 22 06:13:12 UTC 2021
[    6.889863] nvidia_uvm: module uses symbols from proprietary module nvidia, inheriting taint.
[    6.899326] nvidia-uvm: Loaded the UVM driver, major device number 508.
[    6.937538] nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  495.44  Fri Oct 22 06:05:22 UTC 2021
[    6.941599] [drm] [nvidia-drm] [GPU ID 0x00000a00] Loading driver
[    6.941601] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:0a:00.0 on minor 0
[   10.787567] caller _nv000717rm+0x1ad/0x200 [nvidia] mapping multiple BARs
```

Again, downgrading to the 470 drivers resolves these issues.

scaronni, 2021-11-09:
> however any DX11/12 game (via steam, thus Proton) will report an error of `DX11 feature level 10.0 is required to run the engine`.

If it's a regression on Nvidia drivers it needs to be reported upstream.

scaronni, 2021-11-09:
> Nvidia 495.* drivers do not support GTX 980 anymore, so you have to use the 470.* series. I am in the same boat.
> 
> Would it be possible to make 470.* drivers available for Fedora 35? Thanks

I don't have the drivers to test. The request for supporting older drivers came up multiple times, but my resources are finite and if there is no hardware donor I can't support every release with every OS combo, sorry.

combolek, 2021-11-09:
> I don't have the hardware to test. The request for supporting older drivers came up multiple times, but my resources are finite and if there is no hardware donor I can't support every release with every OS combo, sorry.

Fair enough, no need to apologize.

I found the rpm sources in github. Is there a script or some release process to build all of the nvidia stuff?
Thanks

TurboTurtle, 2021-11-09:
> > however any DX11/12 game (via steam, thus Proton) will report an error of `DX11 feature level 10.0 is required to run the engine`.
> 
> If it's a regression on Nvidia drivers it needs to be reported upstream.

Ack, that's fair. Wasn't quite sure if this was a driver regression or something else given the varying behavior.

>The request for supporting older drivers came up multiple times, but my resources are finite and if there is no hardware donor I can't support every release with every OS combo, sorry.

Once I'm able to get a new card (ahaha, right!?) I wouldn't mind donating this current 980, if that'd be useful for you?

mich181189, 2021-12-06:
This might or might not be related, but recently my system has been loading nouveau before nvidia, so nouveau wins and gets control of the card. This wasn't the case before so I'm not sure if I'm just unlucky or if something changed recently

scaronni, 2021-12-09:
> Fair enough, no need to apologize.
>
> I found the rpm sources in github. Is there a script or some release process to build all of the nvidia stuff? Thanks

Just make sure your driver version is in the script + spec file and run `./nvidia-generate-tarball.sh` from the `nvidia-driver` github repo and move on from there.

scaronni, 2021-12-09:
> This might or might not be related, but recently my system has been loading nouveau before nvidia, so nouveau wins and gets control of the card. This wasn't the case before so I'm not sure if I'm just unlucky or if something changed recently

Nothing has changed. No issue on my systems, please open another issue when you have something more specific.

closed 2022-02-03 by scaronni