negativo17/nvidia-driver #117: Hardware acceleration with Wayland

remyabel, 2021-11-09:
This is not really a bug per se...more of a "does anyone else know how to get this working" type of deal. I'm not able to find any concrete information on hardware acceleration on Wayland. These are the relevant packages I have installed.

```
$ rpm -qa | sort | grep vdpau
libva-vdpau-driver-0.7.4-110.fc35.x86_64
libvdpau-1.4-6.fc35.i686
libvdpau-1.4-6.fc35.x86_64
mesa-vdpau-drivers-21.2.5-1.fc35.x86_64
vdpauinfo-1.4-1.fc35.x86_64
$ rpm -qa | sort | grep nvidia
dkms-nvidia-495.44-2.fc35.x86_64
nvidia-driver-495.44-1.fc35.x86_64
nvidia-driver-cuda-495.44-1.fc35.x86_64
nvidia-driver-cuda-libs-495.44-1.fc35.x86_64
nvidia-driver-libs-495.44-1.fc35.i686
nvidia-driver-libs-495.44-1.fc35.x86_64
nvidia-driver-NVML-495.44-1.fc35.x86_64
nvidia-kmod-common-495.44-3.fc35.noarch
nvidia-libXNVCtrl-495.44-1.fc35.x86_64
nvidia-persistenced-495.44-1.fc35.x86_64
nvidia-settings-495.44-1.fc35.x86_64
$ rpm -qa | sort | grep ffmpeg
ffmpeg-4.4.1-1.fc35.x86_64
ffmpeg-devel-4.4.1-1.fc35.x86_64
ffmpeg-libs-4.4.1-1.fc35.x86_64
$ rpm -qa | sort | grep libva
libva-2.13.0-1.fc35.i686
libva-2.13.0-1.fc35.x86_64
libva-utils-2.13.0-1.fc35.x86_64
libva-vdpau-driver-0.7.4-110.fc35.x86_64
$ rpm -qa | sort | grep vaapi
gstreamer1-vaapi-1.19.2-1.fc35.x86_64
```

`vainfo` and `vdpauinfo` do not work:

```
libva info: VA-API version 1.13.0
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
vaInitialize failed with error code -1 (unknown libva error),exit
```

```
display: :1   screen: 0
Xlib:  extension "NV-GLX" missing on display ":1".
Xlib:  extension "NV-GLX" missing on display ":1".
Error creating VDPAU device: 1
```

One can get halfway there for mpv but then the actual hardware acceleration does not work:

```
# This environment variable is required until the upstream bug is fixed
env __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json mpv
# Contents of mpv.conf
profile=gpu-hq                                                                                                                                         
gpu-context=wayland                                                                                                                                    
hwdec=auto-copy-safe
```

This will select the NVIDIA hardware decoder, but hardware acceleration does not actually work (as indicated by CPU usage).

```
[vo/gpu/wayland] GNOME's wayland compositor lacks support for the idle inhibit protocol. This means the screen can blank during playback.
libEGL warning: DRI2: failed to create dri screen
[vo/gpu/opengl] Suspected software renderer or indirect context.
Using hardware decoding (nvdec-copy).
```

The Arch Linux wiki suggests setting `LIBVA_DRIVER_NAME` by looking in `/usr/lib/dri` but there is no `vdpau` library in there. There is one in `/usr/lib/vdpau/` but setting `VDPAU_DRIVER`. Is hardware acceleration in Wayland (**not** XWayland, which is documented to be unsupported) not yet supported or or did I just forget a step somewhere?

leigh123linux, 2021-11-09:
vdpau needs Xorg to run.
TBH nvidia wayland support isn't even beta quality so do yourself a favor and use X instead.

scaronni, 2021-11-09:
> vdpau needs Xorg to run. TBH nvidia wayland support isn't even beta quality so do yourself a favor and use X instead.

Was about to write the same :smile:

closed 2021-11-09 by remyabel