negativo17/nvidia-driver #120: Version 510.47.03 has been released in production branch

mwawra, 2022-02-02:
Hello,

just wanted to mention that 510.47.03 has been released to production branch:
https://www.nvidia.de/Download/driverResults.aspx/186170

Includes support for 3050 and fixing a failure when installed on OS with Linux kernel 5.17 RC.

scaronni, 2022-02-03:
Working on it along with CUDA 11.6.

scaronni, 2022-02-04:
Drivers are going live now with the majority of CUDA 11.6 packages as well.

New driver contains a new service, `nvidia-powerd`. Would be nice to activate this on demand with an if condition in the `systemd` unit to check for the appropriate SBIOS setting: https://download.nvidia.com/XFree86/Linux-x86_64/510.47.03/README/dynamicboost.html

closed 2022-02-04 by scaronni