negativo17/nvidia-driver #121: egl-wayland-1.1.9.el7.x86_64 missing for last update nvidia 510.47.03-1

brunckodusan, 2022-02-04:
Hello,

I would like to turn your attenuation on the last nvidia update for el7. Here 
egl-wayland-1.1.9.el7.x86_64.rpm is missing.

Regards,
                Dusan

joelwampler, 2022-02-04:
I am seeing the same thing in Fedora 34. The latest version of `egl-wayland` on f34 is 1.1.7.. I also checked updates-testing just to be sure and didn't see anything newer.

Thank you!

scaronni, 2022-02-05:
Yes, sorry I pushed the update too quickly, I was not actually finished. New update online in a couple of hours. Thanks.

closed 2022-02-05 by scaronni

scaronni, 2022-02-05:
Should be ok for `el7`, `el8` and `fc34` now:
```
================================================================================
 Package             Arch    Version                     Repository        Size
================================================================================
Installing:
 nvidia-driver       x86_64  3:510.47.03-2.el7           epel-nvidia      2.5 M
Installing for dependencies:
 egl-wayland         x86_64  1.1.7-1.el7                 epel-nvidia       24 k
 kmod-nvidia         x86_64  3:510.47.03-1.el7           epel-multimedia   36 M
 libglvnd-opengl     x86_64  1:1.0.1-0.8.git5baa1e5.el7  base              43 k
 libva-vdpau-driver  x86_64  0.7.4-19.el7                epel              58 k
 nvidia-driver-libs  x86_64  3:510.47.03-2.el7           epel-nvidia      168 M
 nvidia-kmod-common  noarch  3:510.47.03-1.el7           epel-multimedia   20 M

Transaction Summary
================================================================================
Install  1 Package (+6 Dependent packages)
```