negativo17/nvidia-driver #122: Maybe missing egl-gbm dep after latest change?

stevestorey, 2022-02-05:
Hello - many thanks for all the hard work maintaining the driver over the years

I tried upgrading to the nvidia-driver 3:510.47.03-2.fc35 version (where I have both x86_64 and i686 versions of the `nvidia-driver-libs` package installed), which produced some dependency issues:

```
Last metadata expiration check: 0:31:27 ago on Sat 05 Feb 2022 10:57:47 GMT.
Error: 
 Problem: cannot install the best update candidate for package egl-gbm-1.1.0-2.fc35.x86_64
  - cannot install both egl-gbm-1.1.0-1.fc35.x86_64 and egl-gbm-1.1.0-2.fc35.x86_64
  - egl-gbm-1.1.0-1.fc35.i686 has inferior architecture
  - cannot install the best update candidate for package nvidia-driver-libs-3:510.47.03-1.fc35.i686
  - problem with installed package nvidia-driver-libs-3:510.47.03-1.fc35.i686
  - package nvidia-driver-libs-3:510.47.03-2.fc35.i686 requires egl-gbm(x86-32) >= 1.1.0, but none of the providers can be installed
(try to add '--skip-broken' to skip uninstallable packages)
```
so I then upgraded just egl-wayland:
```
# dnf upgrade --exclude=*nvidia* --best --allowerasing
Last metadata expiration check: 0:31:49 ago on Sat 05 Feb 2022 10:57:47 GMT.
Dependencies resolved.
================================================================================================================================================================
 Package                              Architecture                    Version                                  Repository                                  Size
================================================================================================================================================================
Upgrading:
 egl-wayland                          x86_64                          1.1.9-4.fc35                             fedora-multimedia                           36 k
Removing dependent packages:
 egl-wayland                          i686                            1.1.9-3.fc35                             @updates-testing                            70 k

Transaction Summary
================================================================================================================================================================
Upgrade  1 Package
Remove   1 Package

Total download size: 36 k
Is this ok [y/N]: y
```
which then meant I still had issues:
```
dnf upgrade 
Last metadata expiration check: 0:32:01 ago on Sat 05 Feb 2022 10:57:47 GMT.
Dependencies resolved.

 Problem: egl-gbm-1.1.0-1.fc35.i686 has inferior architecture
  - package nvidia-driver-libs-3:510.47.03-2.fc35.i686 requires egl-gbm(x86-32) >= 1.1.0, but none of the providers can be installed
  - cannot install both egl-gbm-1.1.0-1.fc35.x86_64 and egl-gbm-1.1.0-2.fc35.x86_64
  - cannot install the best update candidate for package nvidia-driver-libs-3:510.47.03-1.fc35.i686
  - cannot install the best update candidate for package egl-gbm-1.1.0-2.fc35.x86_64
================================================================================================================================================================
 Package                                  Architecture                 Version                                    Repository                               Size
================================================================================================================================================================
Upgrading:
 nvidia-driver                            x86_64                       3:510.47.03-2.fc35                         fedora-multimedia                       2.6 M
 nvidia-driver-NVML                       x86_64                       3:510.47.03-2.fc35                         fedora-multimedia                       524 k
 nvidia-driver-cuda                       x86_64                       3:510.47.03-2.fc35                         fedora-multimedia                       1.8 M
Skipping packages with conflicts:
(add '--best --allowerasing' to command line to force their upgrade):
 egl-gbm                                  x86_64                       1.1.0-1.fc35                               updates                                  19 k
Skipping packages with broken dependencies:
 nvidia-driver-libs                       i686                         3:510.47.03-2.fc35                         fedora-multimedia                        24 M

Transaction Summary
================================================================================================================================================================
Upgrade  3 Packages
Skip     2 Packages

Total download size: 4.9 M
Is this ok [y/N]: n
Operation aborted
```
In the end, I was able to resolve it by manually installing `egl-gbm.i686` after which the upgrade went through fine:
```
# dnf install egl-gbm.i686
Last metadata expiration check: 0:33:52 ago on Sat 05 Feb 2022 10:57:47 GMT.
Dependencies resolved.
================================================================================================================================================================
 Package                              Architecture                      Version                                      Repository                            Size
================================================================================================================================================================
Installing:
 egl-gbm                              i686                              1.1.0-1.fc35                                 updates                               20 k

Transaction Summary
================================================================================================================================================================
Install  1 Package

Total download size: 20 k
Installed size: 29 k
Is this ok [y/N]: y

# dnf upgrade
Last metadata expiration check: 0:34:11 ago on Sat 05 Feb 2022 10:57:47 GMT.
Dependencies resolved.
================================================================================================================================================================
 Package                                       Architecture               Version                                   Repository                             Size
================================================================================================================================================================
Upgrading:
 nvidia-driver                                 x86_64                     3:510.47.03-2.fc35                        fedora-multimedia                     2.6 M
 nvidia-driver-NVML                            x86_64                     3:510.47.03-2.fc35                        fedora-multimedia                     524 k
 nvidia-driver-NvFBCOpenGL                     x86_64                     3:510.47.03-2.fc35                        fedora-multimedia                      53 k
 nvidia-driver-cuda                            x86_64                     3:510.47.03-2.fc35                        fedora-multimedia                     1.8 M
 nvidia-driver-cuda-libs                       x86_64                     3:510.47.03-2.fc35                        fedora-multimedia                      56 M
 nvidia-driver-devel                           x86_64                     3:510.47.03-2.fc35                        fedora-multimedia                      10 k
 nvidia-driver-libs                            i686                       3:510.47.03-2.fc35                        fedora-multimedia                      24 M
 nvidia-driver-libs                            x86_64                     3:510.47.03-2.fc35                        fedora-multimedia                     168 M

Transaction Summary
================================================================================================================================================================
Upgrade  8 Packages

Total download size: 253 M
```

I'm guessing that `egl-wayland` had a transitive dependency on `egl-gbm` (at least  for i686) and the removal of the egl-wayland dependency needs to be replaced with a dependency on `egl-gbm` ?

scaronni, 2022-02-06:
Unfortunately I synced the mirror by accident before finishing with the packages, which generated in quite a few reports:

closed 2022-02-06 by scaronni

jp7677, 2022-02-06:
The `mutlimedia` repository (https://negativo17.org/repos/multimedia/fedora-35/x86_64/?C=M;O=D) is still behind the `nvidia` (https://negativo17.org/repos/nvidia/fedora-35/x86_64/?C=M;O=D) repository. The sync to `multimedia` will happen automatically at some point?

scaronni, 2022-02-07:
I think you just hit the mirrors in between the syncs, after syncing `nvidia` there was in queue the `multimedia` one.

scaronni, 2022-02-07:
Depending on the stuff I'm syncing it might take 10-15 minutes each.

scaronni, 2022-02-07:
Nevermind, just spot a difference in the repository building scripts, it was not considering `egl-gbm` for multiarch. It will be online in ~15 minutes.

Thanks for reporting!

leigh123linux, 2022-02-07:
@scaronni  Nvidia don't provided 32bit egl-gbm or egl-wayland so why do you?
Does it add any function?

scaronni, 2022-02-07:
Hi Leigh, correct, but Red Hat / Fedora provide it for also i686 so unless it's in the repository it's going to fail on mismatched versions between i686/x86_64.

And this whole Stream thing is just a mess. 8 has old components, 8-stream has new components but it's in the middle of a rebase, 9-stream has latest versions but without the GBM ICD loader yet and no chance to rebuild properly both x86_64 and i686 packages.

I was thinking of shipping all of these myself just to have some sanity control...

jp7677, 2022-02-07:
May be the `Requires` here https://github.com/negativo17/nvidia-driver/blob/master/nvidia-driver.spec#L75 could be also made conditional based on architecture?
This doesn't prevent that the `i686` package needs to exist to prevent mismatched versions, but at least installation of the `i686` package wouldn't be mandatory anymore.

scaronni, 2022-02-07:
> May be the `Requires` here https://github.com/negativo17/nvidia-driver/blob/master/nvidia-driver.spec#L75 could be also made conditional based on architecture? This doesn't prevent that the `i686` [...]

Right, that's a typo, thanks. There was an `%ifnarch` block there that I removed by mistake while doing some cleanup. It should be done for all basic architectures except the i686 libraries:

%ifnarch %{ix86}
%if 0%{?fedora} >= 35 || 0%{?rhel} >= 9
Requires:       egl-gbm%{?_isa} >= 1.1.0
%endif
%endif

stevestorey, 2022-02-07:
Thanks for that - I confirm that having upgraded to `-3` I have now been able to `dnf remove egl-gbm.i686` again without any dependency troubles.