negativo17/nvidia-driver #123: nvidia-driver on el8 requires egl-wayland-1.1.9 which is only available for centos8 stream, not RHEL8

ElCoyote27, 2022-02-06:
Hi,

Just a comment... it seems the latest 510.* rpms were built on CentOS 8 stream.
When I attempted to updated from 495.* to 510.* on RHEL8, the rpms from negativo17 for el8 started requiring egl-wayland >= 1.1.9, which neither EPEL or Centos8 provide.
On RHEL8. the latest (as of Feb 2022) rpms are:
1.1.7-1.el8 for i686 and x86_64.

I was able to work around the issue by downloading these two binary rpms to update my RHEL8 systems:
http://rpmfind.net/linux/centos/8-stream/AppStream/x86_64/os/Packages/egl-wayland-1.1.9-3.el8.i686.rpm http://rpmfind.net/linux/centos/8-stream/AppStream/x86_64/os/Packages/egl-wayland-1.1.9-3.el8.x86_64.rpm

However, since I have not enabled Centos8 Stream on my RHEL8 systems, this bothers me a little bit (downloading 3rd party binary rpms from a rpo which I had never used before).

IMHO, perhaps you could:
1) build and provide these dependencies.
or
2) build on RHEL8 (16 free licences for developpers) or Rocky Linux.

This wasn't a huge blocking point but I thought I'd report it.

ElCoyote27, 2022-02-06:
And thanks for your hard work providing these rpms to the community.

scaronni, 2022-02-06:
Hello, I fixed this yesterday, I synchronized the repositories too quickly: https://github.com/negativo17/nvidia-driver/issues/121

```
$ rpm -qp --requires epel-8/x86_64/nvidia-driver-libs-510.47.03-2.el8.x86_64.rpm | grep egl-wayland
libnvidia-egl-wayland.so.1()(64bit)
[slaanesh@workstation nvidia]$ rpm -qp --provides epel-8/x86_64/egl-wayland-1.1.7-1.el8.x86_64.rpm 
egl-wayland = 1.1.7-1.el8
egl-wayland(x86-64) = 1.1.7-1.el8
libnvidia-egl-wayland.so.1()(64bit)
```
- `epel-8` has `egl-wayland.x86_64` at 1.1.7 (minimum required version).
- `epel-next-8` has `egl-wayland.x86_64` at 1.1.9 since for 8.6 they are rebasing Mesa as well.

scaronni, 2022-02-06:
You can remove `egl-wayland-1.1.9-3.el8.i686.rpm` for now and just run a `dnf distro-sync`.

Nvidia is working closely with Red Hat to make sure all components are in the next RHEL release it seems.

closed 2022-02-06 by scaronni

ElCoyote27, 2022-02-06:
Thank you so much Simone,
It's been great relying on those rpms all those years. Yes, things are 
improving between Red Hat and NVidia but they're still not as good as the 
negativo17 rpms :)

,-*~'`^`'~*-,._.,-*~'`^`'~*-,._.,-*~'`^`'~*-,._.,-*~'`^`'~*-,._.,-*~'`^`'~*-,
Vincent S. Cojot, Computer Engineering. STEP project. _.,-*~'`^`'~*-,._.,-*~
Ecole Polytechnique de Montreal, Comite Micro-Informatique. _.,-*~'`^`'~*-,.
Linux Xview/OpenLook resources page _.,-*~'`^`'~*-,._.,-*~'`^`'~*-,._.,-*~'
http://step.polymtl.ca/~coyote  _.,-*~'`^`'~*-,._ ***@***.***

They cannot scare me with their empty spaces
Between stars - on stars where no human race is
I have it in me so much nearer home
To scare myself with my own desert places.       - Robert Frost