negativo17/nvidia-driver #124: Missing akmod-nvidia.x86_64 version 3:510.47.03-1.fc35 from fedora-nvidia

thestonewell, 2022-02-07:
I saw the updates today for

```
=====================================================================================================================================================================================
 Package                                              Architecture                      Version                                       Repository                                Size
=====================================================================================================================================================================================
Upgrading:
 cuda-cudart                                          x86_64                            1:11.6.55-2.fc35                              fedora-nvidia                            172 k
 nvidia-driver                                        x86_64                            3:510.47.03-3.fc35                            fedora-nvidia                            2.6 M
 nvidia-driver-NVML                                   x86_64                            3:510.47.03-3.fc35                            fedora-nvidia                            524 k
 nvidia-driver-NvFBCOpenGL                            x86_64                            3:510.47.03-3.fc35                            fedora-nvidia                             53 k
 nvidia-driver-cuda                                   x86_64                            3:510.47.03-3.fc35                            fedora-nvidia                            1.8 M
 nvidia-driver-cuda-libs                              x86_64                            3:510.47.03-3.fc35                            fedora-nvidia                             56 M
 nvidia-driver-libs                                   x86_64                            3:510.47.03-3.fc35                            fedora-nvidia                            168 M

```

But I am missing the line `akmod-nvidia.x86_64    3:510.47.03-3.fc35`. Only `3:510.47.03-1.fc35` is available.

Not sure it's a sync issue like the ones we saw the last two days. Is the package just missing???

- Thomas

scaronni, 2022-02-08:
Completely unrelated, it's a separate package, the requirement is only on the version, not the release:
```
$ rpm -qp --requires x86_64/nvidia-driver-510.47.03-3.fc35.x86_64.rpm | grep kmod
nvidia-kmod-common = 3:510.47.03
$ rpm -qp --requires x86_64/nvidia-kmod-common-510.47.03-1.fc35.noarch.rpm | grep kmod
config(nvidia-kmod-common) = 3:510.47.03-1.fc35
nvidia-kmod = 3:510.47.03
```

closed 2022-02-08 by scaronni

thestonewell, 2022-02-08:
Well, it being dependent just on the release is certainly legit. Keeping the akmods version (in this case -3) in sync though helps big time to get the right updates. I have rpm-fusion in parallel installed. Not having the -3 here would cause the rpm-fusion builds to come in. Example if I were not careful:

```
=====================================================================================================================================================================================
 Package                                            Architecture                  Version                                     Repository                                        Size
=====================================================================================================================================================================================
Upgrading:
 akmod-nvidia                                       x86_64                        3:510.47.03-2.fc35                          rpmfusion-nonfree-updates                         28 k
 cuda-cudart                                        x86_64                        1:11.6.55-2.fc35                            fedora-nvidia                                    172 k
 nvidia-driver                                      x86_64                        3:510.47.03-3.fc35                          fedora-nvidia                                    2.6 M
 nvidia-driver-NVML                                 x86_64                        3:510.47.03-3.fc35                          fedora-nvidia                                    524 k
 nvidia-driver-NvFBCOpenGL                          x86_64                        3:510.47.03-3.fc35                          fedora-nvidia                                     53 k
 nvidia-driver-cuda                                 x86_64                        3:510.47.03-3.fc35                          fedora-nvidia                                    1.8 M
 nvidia-driver-cuda-libs                            x86_64                        3:510.47.03-3.fc35                          fedora-nvidia                                     56 M
 nvidia-driver-libs                                 x86_64                        3:510.47.03-3.fc35                          fedora-nvidia                                    168 M
Installing dependencies:
 xorg-x11-drv-nvidia-kmodsrc                        x86_64                        3:510.47.03-1.fc35                          rpmfusion-nonfree-updates                         28 M

Transaction Summary
=====================================================================================================================================================================================
```

So while not strictly needed to rev from a functionality point of view I still recommend it for usability purposes. Thoughts?

leigh123linux, 2022-02-08:
> So while not strictly needed to rev from a functionality point of view I still recommend it for usability purposes. Thoughts?

Don't use conflicting repos!, if you do try adding a nvidia exclude to the rpmfusion repo file.