negativo17/nvidia-driver #125: Replace libva-vdpau-driver requirement with newer nvidia-vaapi-driver

jp7677, 2022-02-12:
Hi,

May be it is still to early, but it would be cool to change the required dependency to `libva-vdpau-driver` (https://github.com/negativo17/nvidia-driver/blob/master/nvidia-driver.spec#L44) at some point to the newer `nvidia-vaapi-driver` (https://github.com/elFarto/nvidia-vaapi-driver). This newer VAAPI driver uses NVDEC directly instead of the now older VDPAU interface and seems to work nicely together with Firefox.

Kind regards

scaronni, 2022-02-12:
Hi, I was actually looking at it a few days ago. It will probably only work with newer `libva` versions, so it might not be for all branches.

This reminded me of removing the forced dependency on `libva-vdpau-driver` which is actually not working properly anymore.

jp7677, 2022-02-12:
> Hi, I was actually looking at it a few days ago. It will probably only work with newer `libva` versions, so it might not be for all branches.

Right, yes, I didn't consider all OS's, I had indeed only recent Fedora in mind ;).

> This reminded me of removing the forced dependency on `libva-vdpau-driver` which is actually not working properly anymore.

Yeah, that's I guess a wise thing to do.

scaronni, 2022-02-12:
The naming part of the project is still a bit confusing:
- https://forums.developer.nvidia.com/t/nvidia-va-api-implementation-v0-0-1-release/199437/2
- https://github.com/intel/libva/pull/554

For now I've just removed the hard dependency on `nvidia-vdpau-driver` for `nvidia-driver` and pushed `nvidia-vaapi-driver` to the repositories. It's available only on Fedora due to a Meson dependency. Considering it's only using `nv-codec-headers` it could be in Fedora as well.

Packages will be online soon. To use them, update the Nvidia driver packages and then run `dnf swap libva-vdpau-driver nvidia-vaapi-driver`.
```
$ vainfo
error: can't connect to X server!
libva info: VA-API version 1.13.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
```

elFarto, 2022-02-12:
Hey, just noticed you've package my driver up. I actually only noticed it as I had an old test RPM installed that I created that had exactly the same name and it thought I needed to update :smile:.

I'm not sure putting that script in the profile.d directory is the best idea. The MOZ_DISABLE_RDD_SANDBOX variable weakens the security of Firefox around decoding video. And the LIBVA_DRIVER_NAME forces it to use the driver, when you could be using a hybrid Intel/NVIDIA system and you want to use the other one (and there are other problems if you try and decode video on NVIDIA and export it to Intel). Both the variables are still needed, and likely will be for some time.

I think it would be worth at least noting what the package does somewhere, just so users don't get tripped up by it.

jp7677, 2022-02-12:
@elFarto First and foremost thanks a lot for all your hard work! Out of curiosity, what are exactly the consequence of disabling the RDD sandbox? I know what a sandbox is, but I'm not familiar with what exactly is sand-boxed in this case and what could potentially go wrong if not. Also, is this only needed for your driver or for all VAAPI drivers?

scaronni, 2022-02-13:
@elFarto thank you very much for your work!

> I'm not sure putting that script in the profile.d directory is the best idea. The MOZ_DISABLE_RDD_SANDBOX variable weakens the security of Firefox around decoding video. And the LIBVA_DRIVER_NAME [...]

I think I will just remove the profile file and include the `libva` patch (in my multimedia repository) that @elFarto created for the detection until upstream decides what to do. This would give me the opportunity to enable it as well for the CentOS releases where I provided an updated `libva` along with all the Intel stack.

This is also in a third party repository that users explicitly enable themselves, so I think I'm pretty free here.

@jp7677 are you using my multimedia repository?

jp7677, 2022-02-13:
@scaronni yes, I’m using the multimedia repository. Usually this one only, but for e.g. yesterday I additionally enabled the nvidia repo for having to wait until the multimedia repo is in sync.

scaronni, 2022-02-13:
They're in sync, i usually push to both (~10 minutes delay) except some rare circumstances.

jp7677, 2022-02-13:
@scaronni I just received your latest version. As expected, I have to put `export MOZ_DISABLE_RDD_SANDBOX=1` in my `.bash_profile` to get it to work. Setting `LIBVA_DRIVER_NAME` is, also as expected, no longer needed.

Really cool, thanks a lot! Now waiting for the Firefox 98 release in March for AV1 support ;). Very impressive to move from zero hardware acceleration to complete support in just a few month time! I guess the work here is done.

closed 2022-02-13 by jp7677