negativo17/nvidia-driver #127: Missing egl-gbm package in epel-9 repository

kmittman, 2022-03-28:
Hi @scaronni,
It appears the `nvidia-driver-libs` package depends on [egl-gbm](https://github.com/negativo17/egl-gbm) but no such RPM exists in the [negativo17/epel-9](https://negativo17.org/repos/nvidia/epel-9/x86_64/) repository. I do see it in the [negativo17/fedora35](https://negativo17.org/repos/nvidia/fedora-35/x86_64/egl-gbm-1.1.0-2.fc35.x86_64.rpm) repository.

Steps to reproduce:
```
$ docker run -it quay.io/centos/centos:stream9
echo -e '[negativo17-epel9]\nname=negativo17-epel9' > /etc/yum.repos.d/negativo17-epel9.repo
echo -e 'enabled=1\ngpgcheck=1' >> /etc/yum.repos.d/negativo17-epel9.repo
echo 'baseurl=https://negativo17.org/repos/nvidia/epel-9/x86_64/' >> /etc/yum.repos.d/negativo17-epel9.repo
echo 'gpgkey=https://negativo17.org/repos/RPM-GPG-KEY-slaanesh' >> /etc/yum.repos.d/negativo17-epel9.repo
dnf install nvidia-driver
```
```
CentOS Stream 9 - BaseOS                                                    3.7 MB/s | 5.6 MB     00:01    
CentOS Stream 9 - AppStream                                                 4.7 MB/s |  12 MB     00:02    
CentOS Stream 9 - Extras packages                                           9.3 kB/s | 6.8 kB     00:00    
negativo17-epel9                                                             57 kB/s |  93 kB     00:01    
Error: 
 Problem: package nvidia-driver-3:510.60.02-1.el9.x86_64 requires libnvidia-glcore.so.510.60.02()(64bit), but none of the providers can be installed
  - package nvidia-driver-3:510.60.02-1.el9.x86_64 requires libnvidia-tls.so.510.60.02()(64bit), but none of the providers can be installed
  - package nvidia-driver-3:510.60.02-1.el9.x86_64 requires nvidia-driver-libs(x86-64) = 3:510.60.02, but none of the providers can be installed
  - cannot install the best candidate for the job
  - nothing provides egl-gbm(x86-64) >= 1.1.0 needed by nvidia-driver-libs-3:510.60.02-1.el9.x86_64
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)
```

Note: this is not a package we currently produce in the CUDA repository, though [given this line](https://github.com/NVIDIA/yum-packaging-nvidia-driver/blob/main/nvidia-generate-tarballs.sh#L28) `rm -f libnvidia-egl-gbm.so.*` that may be an oversight; however, we don't have the _Requires:_ in our .spec

scaronni, 2022-03-29:
Hi Kevin!

I'm not yet finished with the `epel-9` repository, I just bumped up yesterday by adding all the `i686` packages, but haven't tested yet. The package is missing, indeed, it will be online shortly.

It is not an oversight as you're publishing it as open source [1], so as part of the build I'm using as much open source components as possible.

[1] https://github.com/NVIDIA/egl-gbm

scaronni, 2022-03-29:
Packages are online.

closed 2022-03-29 by scaronni