negativo17/nvidia-driver #128: libnvidia-vulkan-producer.so symlink is missing

flibitijibibo, 2022-05-04:
This was discovered as part of SDL's migration to Wayland by default: https://github.com/libsdl-org/SDL/issues/5527#issuecomment-1101442688

For whatever reason it seems the driver _really_ wants there to be an unversioned .so that links to the fully-versioned file. This is done in the Flatpak version of the driver, for example. Without this, we're unable to create Vulkan surfaces on Wayland natively (though interestingly Xwayland still works).

This can be tested with vulkaninfo on a Wayland desktop: It will currently crash, but adding the symlink makes it work again.

scaronni, 2022-05-15:
Thanks for reporting and sorry for the delay. Will probably come along with the latest driver today.

I have many of your ports, keep up the great work!! :pray:

closed 2022-05-15 by scaronni

scaronni, 2022-05-15:
Had to reshuffle things around a bit (rewriting history) as the symlink alone is not enough, otherwise I would have needed to add a filter as well which I'm trying to avoid: https://github.com/negativo17/nvidia-driver/commit/f8b9eea2cc55850050edc50176947b5c44bcf04f