negativo17/nvidia-driver #129: Fedora 36 gdm green screen

jp7677, 2022-05-12:
Hi there,

I've updated two days ago from Fedora 35 to 36. I did so by first remove the NVIDIA driver, then performing the upgrade using nouveau and then re-installing the NVIDIA driver from here again. After rebooting I was greeted by a completely green login screen. Eventually I figured that I could login blindly. Doing some research it seems I run into https://forums.developer.nvidia.com/t/510-39-01-on-5-16-0-kernel-green-screen/200476/5 and the reason seems to be that FC36 enabled simpledrm (I'm just repeating what I read, I'm not familiar with the details). I could fix it with adding `initcall_blacklist=simpledrm_platform_driver_init` to the kernel boot arguments as stated in that post.

Is this supposed to happen and is this the correct fix?

Interesting note is that I added that boot argument only for a specific kernel by executing `sudo grubby --update-kernel /boot/vmlinuz-5.17.5-300.fc36.x86_64 --args="initcall_blacklist=simpledrm_platform_driver_init"`, but even with a kernel update in the meantime to 5.17.6, the boot argument is still there.

Kind regards.

thestonewell, 2022-05-13:
I've moved to rpmfusion for now. They have applied patches as per https://github.com/rpmfusion/nvidia-kmod to overcome the simplerpm issue for now.

Should we consider the same patches?

jp7677, 2022-05-13:
Thanks for your response. I cannot judge the patches itself, but yeah, including those two here sounds like a possible way forward. Based on what I see in the rpmfusion repo, those patches are relevant for both the 510 and 515 driver version.

For my understanding, is everyone with Fedora 36 and drivers from here affected by this, or do I hit an edge case somewhere?

cc @scaronni since I saw your commits for the 515 driver.

vpxyz, 2022-05-14:
Same problem here  (with sddm) but I see a black screen.  I have a Xeon cpu without integrated Intel video card and with just one monitor. I'm not sure but I think sddm show login screen on a ghost monitor and a black screen on my monitor 
xrandr --verbose show a ghost monitor:

`
None-1-1 connected (normal left inverted right x axis y axis)
        Identifier: 0x25b
        Timestamp:  29085
        Subpixel:   unknown
        Clones:    
        CRTCs:      4
        Transform:  1.000000 0.000000 0.000000
                    0.000000 1.000000 0.000000
                    0.000000 0.000000 1.000000
                   filter: 
        PRIME Synchronization: 1 
                supported: 0, 1
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 31 
                supported: 31
        non-desktop: 0 
                range: (0, 1)
  1280x1024 (0x25d) 78.643MHz +preferred
        h: width  1280 start 1280 end 1280 total 1280 skew    0 clock  61.44KHz
        v: height 1024 start 1024 end 1024 total 1024           clock  60.00Hz
 `
I can log in (by simply typing in a password, blindly) and everything works fine.

jp7677, 2022-05-14:
Hi @vpxyz yeah, this looks like the same issue, I also saw a ghost monitor next to my actual monitor after having logged in blindly.

stevestorey, 2022-05-14:
> is everyone with Fedora 36 and drivers from here affected by this,

I believe so - I certainly am. Technically the issue is within the `nvidia-kmod` repo rather than the driver one - I have made a possible PR here: https://github.com/negativo17/nvidia-kmod/pull/12 which works for me, and boots to sddm & plasma for me.

It doesn't however fix the fact that the Plymouth boot screen and TTYs (Ctrl-Alf-F3 and so on) don't work. Those points would need nvidia to make more substantive changes in order for the driver to work properly with `simplefb` instead of `efifb` (as per https://bugzilla.redhat.com/show_bug.cgi?id=2071209 ). Working around thos for the mo can be done by building your own kernel with the `simpledrm` config changes reverted, and going back to the `efifb` instead.

scaronni, 2022-05-15:
I'm using Gnome/GDM on Fedora 36 and I don't see any issue at all with the driver as is, neither in 510 nor in 515.

@jp7677 are you using something else than GDM like everybody else here seems to?

scaronni, 2022-05-15:
> I believe so - I certainly am. Technically the issue is within the `nvidia-kmod` repo rather than the driver one - I have made a possible PR here: [negativo17/nvidia-kmod#12](https://github.com/negativo17/nvidia-kmod/pull/12) which works for me, and boots to sddm & plasma for me.

Both patches check for `CONFIG_SYSFB_SIMPLEFB` and `CONFIG_DRM_SIMPLEDRM` being enabled, so it seems it could be applied unconditionally, but `drm/drm_aperture.h` is only available on CentOS 8+.

Your merge request has enabled these patches unconditionally, while I'm trying to build (or allow building) the same release on every distribution.

scaronni, 2022-05-15:
I'll just add a conditional for now as well (packaging policies say that patches should not be selectively applied depending on various conditions, so I'm not really in favor of that).

steve-tg, 2022-05-15:
> I'm using Gnome/GDM on Fedora 36 and I don't see any issue at all with the driver as is, neither in 510 nor in 515.

Interesting. Speaking for myself, I use sddm + plasma, both on X11 rather than Wayland. I don't technically see a green screen, I get nothing at all. Do your TTYs work as well or not ?

> Your merge request has enabled these patches unconditionally, while I'm trying to build (or allow building) the same release on every distribution.
> I'll just add a conditional for now as well (packaging policies say that patches should not be selectively applied depending on various conditions, so I'm not really in favor of that).

Ah right, sorry - I hadn't appreciated that detail, thanks! Prob worth highlighting that I'm going down the "rebuild the kernel" route so that I can get the ttys working again, in case of needing to debug issues etc.

stevestorey, 2022-05-15:
> I'm using Gnome/GDM on Fedora 36 and I don't see any issue at all with the driver as is, neither in 510 nor in 515.

Interesting. Speaking for myself, I use sddm + plasma, both on X11 rather than Wayland. I don't technically see a green screen, I get nothing at all. Do your TTYs work as well or not ?

> Your merge request has enabled these patches unconditionally, while I'm trying to build (or allow building) the same release on every distribution.
> I'll just add a conditional for now as well [...]

Ah right, sorry - I hadn't appreciated that detail, thanks! Prob worth highlighting that I'm going down the "rebuild the kernel" route so that I can get the ttys working again, in case of needing to debug issues etc.

scaronni, 2022-05-15:
> I've updated two days ago from Fedora 35 to 36. I did so by first remove the NVIDIA driver, then performing the upgrade using nouveau and then re-installing the NVIDIA driver from here again.

I normally update without removing anything.

scaronni, 2022-05-15:
> Interesting. Speaking for myself, I use sddm + plasma, both on X11 rather than Wayland. I don't technically see a green screen, I get nothing at all. Do your TTYs work as well or not ?

Absolutely, no issues at all. I mean everything works as in Fedora 35.

thestonewell, 2022-05-15:
The issue is totally reproducible IF efifb is disabled in the kernel as it is right now in the fc36 stock kernel.

And yes, if you see a green screen or not probably depends on the age of your hardware. My old box takes about 10sec to boot.
- see the grub for 1s (my local setting)
- during plymout you see a green screen
- during sddm (or gdm) you see a black screen (aka nothing)

Now if your default sessions logs you into Gnome (both Wayland and X11) you will now see in your display settings a 2nd monitor even if you just have one. This is the simpledrm one. Trying to disable that one will likely crash you session if you on X11. With Wayland it doesn't.

If you log into KDE I did not even get a session running on Wayland. For X11 it works yet this second monitor exists too and shows in nvidia-settings as prime monitor. You need to be careful to not place any content on this screen or even move the mouse to it. It will crash the X11 session.

I tried for two long nights :)

@scaronni, can you can also check via `systool -c drm -p` or `ls -l /sys/class/drm/card*` you see two cards even though you just have one physical. In case the patch is not applied on the stock kernel you have car0 backed by simpledrm

as said in https://github.com/negativo17/nvidia-driver/issues/129#issuecomment-1126706745 though the patch does give us a display back but since it disables simpledrm and the stock kernel has no efifb anymore there is no TTY to use :(

HTH

jp7677, 2022-05-15:
> I'm using Gnome/GDM on Fedora 36 and I don't see any issue at all with the driver as is, neither in 510 nor in 515.
> 
> @jp7677 are you using something else than GDM like everybody else here seems to?

I’m running pretty much vanilla Fedora 36 with gdm/Gnome and only your multimedia repo as additional repo, although it is an update from FC35, thus no clean installation.

Are you running a hybrid setup by chance, if yes, may be that masks this issue somehow? I’m just running a single discrete GPU (Ampere).

scaronni, 2022-05-15:
Thanks for the information, to be honest I noticed the second monitor just now, it did not cause any issue for me so far.

The patches, disabling simpledrm, also disable printing anything if you press "ESC" during the graphical boot to see if the system is doing anything. I don't think they are viable right now, considering GDM/Gnome don't have any issue beside the second monitor.

This black screen thing on SDDM reminds me of the original issues with the Optimus laptops, there was something very similar to now. Can you configure SDDM to use the Nvidia screen instead of the simpledrm one? If yes, no patches or custom kernel is needed, you can still see the console (at boot, tty switching) and you just have the annoying second screen.

scaronni, 2022-05-15:
> I’m running pretty much vanilla Fedora 36 with gdm/Gnome and only your multimedia repo as additional repo, although it is an update from FC35, thus no clean installation.
>
> Are you running a hybrid [...]

No exactly the same setup as you.

scaronni, 2022-05-15:
Shall we publish the patches as well? Not sure about the result, I fear tons of people will write me that the console does not work anymore.

scaronni, 2022-05-15:
@jp7677 does adding the boot parameter fix the green screen you're having but as well get rids of the second simpledrm screen once booted and tty switching works still fine?

scaronni, 2022-05-15:
I will make an attempt without patches and just the boot option: https://forums.developer.nvidia.com/t/510-39-01-on-5-16-0-kernel-green-screen/200476/17

stevestorey, 2022-05-15:
> The patches, disabling simpledrm, also disable printing anything if you press "ESC" during the graphical boot to see if the system is doing anything. I don't think they are viable right now, considering GDM/Gnome don't have any issue beside the second monitor.

I had expected that the boot messages indeed wouldn't show - but I'm surprised that they're showing for you at all as I don't get anything either with, or without the patches (hence why I thought that 'with' was a better outcome as you at least get the graphical boot to work). Are booting UEFI or legacy (I'm UEFI, and as a result of Fedora removing `efifb` driver from the stock kernel, I believe there's no way that TTYs nor pressing escape during boot would work - but reading all the  stuff above, I'm now more confused about what might be going on)

jp7677, 2022-05-15:
> @jp7677 does adding the boot parameter fix the green screen you're having but as well get rids of the second simpledrm screen once booted and tty switching works still fine?

The boot parameter (`initcall_blacklist=simpledrm_platform_driver_init`) fixes the green screen for me and also I'm no longer seeing two displays. Though no tty with it. I just copied that parameter from the NVIDIA forum, so not sure about other consequences anywhere else.
For the record, I'm also on UEFI.

scaronni, 2022-05-15:
Lol, I think these patches are a bit too rushed. So summing up, at least for my setup:
- Upgraded Fedora to 36 (doesn't really matter)
- GDM/Gnome with default Wayland on GDM
- Nvidia driver 515.43.04 as a test
- Without patches: 2nd Simple DRM display, everything works fine
- With patches: no 2nd Simple DMR display, no simple drm at all no output when TTY switching, no console output when pressing ESC in Plymouth

scaronni, 2022-05-15:
It seems 515.43.04 at least makes the system start without any green screen or whatsoever.
I will push a plain update to 515.43.04 (`nvidia-settings` also works under Wayland now) and then make some more tests.

jp7677, 2022-05-15:
Really strange situation and I'm also starting to get confused. 
I mean, I would have at least expected more info on the usual sites that FC36 with NVIDIA either misses tty (when using rpmfusion) or has issues with GDM or other login manager showing up at all when installing the drivers manually or from here. But so far internet search doesn't show much.

leigh123linux, 2022-05-15:
> Lol, I think these patches are a bit too rushed. So summing up, at least for my setup:
> 
> * Upgraded Fedora to 36 (doesn't really matter)
> * GDM/Gnome with default Wayland on GDM
> * Nvidia driver 515.43.04 as a test
> * Without patches: 2nd Simple DRM display, everything works fine
> * With patches: no 2nd Simple DRM display, no simple drm at all no output when TTY switching, no console output when pressing ESC in Plymouth

Are you using pure UEFI or UEFI with CSM enabled?

scaronni, 2022-05-15:
Pure UEFI. Do you think CSM enabled or not would make a difference?

scaronni, 2022-05-15:
> Really strange situation and I'm also starting to get confused. I mean, I would have at least expected more info on the usual sites that FC36 with NVIDIA either misses tty (when using rpmfusion) or has issues with GDM or other login manager showing up at all (or at least showing a second display) when installing the drivers manually or from here. But so far internet search doesn't show much.

Well, there's also this one: https://github.com/negativo17/nvidia-driver/commit/f8b9eea2cc55850050edc50176947b5c44bcf04f

The fix is not to create the symlink as suggested by Nvidia, but it would be to provide the correct `SONAME` in the library, and then at that point it can even be renamed and it would not matter much as `ldconfig` would create the correct link.

To be honest, I'm not even bothering with Wayland on Nvidia at this time (that's why I did not spot it), there are rough edges everywhere.

scaronni, 2022-05-15:
@stevestorey @jp7677 I'm about to push the new beta driver to the Fedora 36 repository *without any patch*, please let me know if you see any improvements as is. Thanks.

I've moved the patches to the `simpledrm` branch of `nvidia-kmod` and `dkms-nvidia`.

leigh123linux, 2022-05-15:
> Pure UEFI. Do you think CSM enabled or not would make a difference?

I was curious to why you don't have the issue.
My hardware is affected.

```
$ inxi -SGM
System:
  Host: fedora Kernel: 5.17.5-300.fc36.x86_64 arch: x86_64 bits: 64
    Desktop: Cinnamon v: 5.2.7 Distro: Fedora release 36 (Thirty Six)
Machine:
  Type: Desktop System: CCL s
    product: Customised AMD Zen Bundles Motherboard Bundle v: N/A
    serial: <superuser required>
  Mobo: ASUSTeK model: ROG STRIX B450-F GAMING v: Rev 1.xx
    serial: <superuser required> UEFI: American Megatrends v: 4801
    date: 03/02/2022
Graphics:
  Device-1: NVIDIA TU117 [GeForce GTX 1650] driver: nvidia v: 515.43.04
  Display: x11 server: X.Org v: 1.20.14 driver: X: loaded: nvidia
    unloaded: fbdev,modesetting,nouveau,vesa gpu: nvidia,nvidia-nvswitch
    resolution: 3840x2160~60Hz
  OpenGL: renderer: NVIDIA GeForce GTX 1650/PCIe/SSE2
    v: 4.6.0 NVIDIA 515.43.04
```

I tried GDM and lightdm without the patches, both were broken (black screen).

scaronni, 2022-05-15:
I just tested an Optimus laptop from work and it all works fine, as the simpledrm is bound to the main display as well on Intel.

@leigh123linux where does this patch come from?

https://pkgs.rpmfusion.org/cgit/nonfree/nvidia-kmod.git/tree/nvidia-kmod-simpledrm.patch

It says "[1/2]", I would be curious to see what's left.

leigh123linux, 2022-05-15:
The patches were supplied the change owner.

https://fedoraproject.org/wiki/Changes/ReplaceFbdevDrivers

Here's the unused  patch

```
From 3ae2c20e626c3031aef64e5d8636a5d555c71c83 Mon Sep 17 00:00:00 2001
From: Javier Martinez Canillas <javierm@redhat.com>
Date: Fri, 18 Feb 2022 13:48:47 +0100
Subject: [PATCH 2/2] nvidia:-drm: setup generic fbdev emulation support

There were reports that when using the simpledrm driver to have an early
framebuffer, the nvidia driver doesn't have VT consoles when takes over.

This seems to be due not registering the DRM generic fbdev emulation and
that leads to fbcon not finding an fbdev to use as a framebuffer console.

---
 kernel/nvidia-drm/nvidia-drm-drv.c | 7 +++++++
 1 file changed, 7 insertions(+)

diff --git a/kernel/nvidia-drm/nvidia-drm-drv.c b/kernel/nvidia-drm/nvidia-drm-drv.c
index 9149f87b6244..61a01083845d 100644
--- a/kernel/nvidia-drm/nvidia-drm-drv.c
+++ b/kernel/nvidia-drm/nvidia-drm-drv.c
@@ -61,6 +61,7 @@
 #endif
 
 #include <drm/drm_aperture.h>
+#include <drm/drm_fb_helper.h>
 
 #include <linux/pci.h>
 
@@ -958,6 +959,12 @@ static void nv_drm_register_drm_device(const nv_gpu_info_t *gpu_info)
         goto failed_drm_register;
     }
 
+    if (IS_ENABLED(CONFIG_DRM_SIMPLEDRM)) {
+	    /* Setup generic fbdev emulation  */
+
+	    drm_fbdev_generic_setup(dev, 32);
+    }
+
     /* Add NVIDIA-DRM device into list */
 
     nv_dev->next = dev_list;
-- 
2.34.1
```

jp7677, 2022-05-15:
@scaronni Unfortunately I'm seeing the same green (and then black) GDM screen and the second simpledrm monitor with driver version 515. The initial reboot was fine because I still had `initcall_blacklist=simpledrm_platform_driver_init` applied, but once I remove that and restarted I got the initial broken behavior.

jp7677, 2022-05-15:
PPS: Looking at the display in Gnome settings, the simpledrm display is listed as display 1 and my actual monitor is display 2, I would assume this is/was reversed for you?

jp7677, 2022-05-15:
Not sure if this is relevant, but I'm using dkms instead of akmod on Fedora. (sry, just remembered just now that this is not the default on Fedora)

scaronni, 2022-05-15:
> The patches were supplied the change owner.
> 
> https://fedoraproject.org/wiki/Changes/ReplaceFbdevDrivers
> 
> Here's the unused  patch
> 
> ```
> From 3ae2c20e626c3031aef64e5d8636a5d555c71c83 Mon Sep 17 00:00:00 2001
> From: Javier Martinez Canillas <javierm@redhat.com>
> Date: Fri, 18 Feb 2022 13:48:47 +0100
> Subject: [PATCH 2/2] nvidia:-drm: setup generic fbdev emulation support
> 
> There were reports that when using the simpledrm driver to have an early
> framebuffer, the nvidia driver doesn't have VT consoles when takes over.
> 
> This seems to be due not registering the DRM generic fbdev emulation and
> that leads to fbcon not finding an fbdev to use as a framebuffer console.
> 
> Signed-off-by: Javier Martinez Canillas <javierm@redhat.com>
> ---
>  kernel/nvidia-drm/nvidia-drm-drv.c | 7 +++++++
>  1 file changed, 7 insertions(+)
> 
> diff --git a/kernel/nvidia-drm/nvidia-drm-drv.c b/kernel/nvidia-drm/nvidia-drm-drv.c
> index 9149f87b6244..61a01083845d 100644
> --- a/kernel/nvidia-drm/nvidia-drm-drv.c
> +++ b/kernel/nvidia-drm/nvidia-drm-drv.c
> @@ -61,6 +61,7 @@
>  #endif
>  
>  #include <drm/drm_aperture.h>
> +#include <drm/drm_fb_helper.h>
>  
>  #include <linux/pci.h>
>  
> @@ -958,6 +959,12 @@ static void nv_drm_register_drm_device(const nv_gpu_info_t *gpu_info)
>          goto failed_drm_register;
>      }
>  
> +    if (IS_ENABLED(CONFIG_DRM_SIMPLEDRM)) {
> +	    /* Setup generic fbdev emulation  */
> +
> +	    drm_fbdev_generic_setup(dev, 32);
> +    }
> +
>      /* Add NVIDIA-DRM device into list */
>  
>      nv_dev->next = dev_list;
> -- 
> 2.34.1
> ```

This seems exactly to enable the fbdev emulation on nvidia-drm to have the console after the first patch.

Any specific reason why this was not added to rpmfusion?

scaronni, 2022-05-15:
> PPS: Looking at the display section in Gnome settings, the simpledrm display is listed as display 1 and my actual monitor is display 2, I would assume this is/was reversed for you? 

No, i have the same, the simpledrm one is the first.

scaronni, 2022-05-15:
I will make some tests with both patches and report here.

stevestorey, 2022-05-15:
>  I will make some tests with both patches and report here.

I will do the same, but it won't be until tomorrow I'm afraid (UK time).

> I've moved the patches to the simpledrm branch of nvidia-kmod and dkms-nvidia.

Yep, fine - given not everyone is experiencing the problem, I agree it's better to try and understand it better before forcing patches on everyone.

> I just tested an Optimus laptop from work and it all works fine

Are all your systems Optimus, and are they all working - is that the crucial difference? - I also have a single discrete 2060 as the primary GPU, and I _think_, so does everyone else where with issues.

Does anyone here either have issues with an Optimus laptop, or not have issues with a discrete card ?

scaronni, 2022-05-15:
Worse than ever. Same as before, but the moment I try to go to a console the system just hangs with three blue small dots in the upper left corner. I can't test also what happens on an Optimus laptop (colleague has left with the laptop).

In the case of this Optimus configuration, the simpledrm device is bound to the main screen driven by an intel driver, so I guess that even if it works, the second card (Nvidia) takes over and can never go into standby as it's rendering the main screen.

If you want to try yourself here are the packages to install manually:

https://negativo17.org/repos/nvidia_issue_129/akmod-nvidia-515.43.04-2.fc36.x86_64.rpm
https://negativo17.org/repos/nvidia_issue_129/dkms-nvidia-515.43.04-2.fc36.x86_64.rpm

jp7677, 2022-05-15:
> If you want to try yourself here are the packages to install manually:
> 
> https://negativo17.org/repos/nvidia_issue_129/akmod-nvidia-515.43.04-2.fc36.x86_64.rpm https://negativo17.org/repos/nvidia_issue_129/dkms-nvidia-515.43.04-2.fc36.x86_64.rpm

Using this version I'm also having no tty and once I try to switch to a tty (e.g. alt+f3) I cannot go back to the graphical console :(, Although I didn't see anything on the tty earlier (monitor said no signal then), I'm pretty certain that switching back had worked.

Everything else though, works fine, including suspend/resume which was unreliable earlier.

msmafra, 2022-05-15:
I just had the green screen, black screen, no TTY nor being able to use the terminal after a minute or so to try to remove the drivers on my laptop on dGPU nor Hybrid mode (which I can change on the UEFI). Had the same problems with Fedora 36 during all the testing (even two days before the BETA).

scaronni, 2022-05-16:
With the kernel boot parameter and no patches I'm in the same situation as the driver disabling simpledrm; 2nd monitor disappears but no consoles at all.

scaronni, 2022-05-16:
I made some tests with a Lenovo P52 at work, that's an Optimus laptop with an Intel driving the main screen and a Quadro P500 as a standby discrete card, with no outputs directly connected. Everything work as is with the vanilla driver (fb device on simpledrm with the intel card, Nvidia driver on the second card only), I see just one screen.

System locks up completely with the patches (with and without the second one for generating the fb emulation).

I'm not putting the patches in. It seems the best approach for people with the issue is the boot parameter, which gives the same results as the patch to take over the simpledrm driver (i.e. no console, no 2nd fake screen).

jp7677, 2022-05-16:
I'll test a bit more with suspend/resume since it was getting late yesterday. I had the idea that suspend/resume worked perfectly with the patches applied, but was unreliable with just the boot parameter (e.g. sometimes my machine doesn't went to suspend or immediately came back from suspend). Would be nice if others could collect and note their findings here about suspend/resume behavior.

Anyway, this is a horrible situation with no good solution and I really think now that the upstream decision to go for simpledrm came to earlier :(.

scaronni, 2022-05-17:
I agree, but what can I say...

martinezjavier, 2022-05-19:
I've proposed the following workaround https://gitlab.com/cki-project/kernel-ark/-/merge_requests/1788.

TL; DR the kernel will choose to either register a platform device that binds to `simpledrm` or `{efi,vesa}fb` depending on whether `nvidia-drm.modeset=1` is set or not.

scaronni, 2022-05-22:
Interesting.
I think this will be fixed by Nvidia in the next 515 series driver, anyway.

What is kernel-ark? I'm reading the docs but besides guidelines and contribution instructions it does not say what's its purpose.

xvitaly, 2022-05-22:
>  I think this will be fixed by Nvidia in the next 515 series driver, anyway.

But it will never be fixed in legacy branches (470.xx, 390.xx).

scaronni, 2022-05-22:
> > I think this will be fixed by Nvidia in the next 515 series driver, anyway.
> 
> But it will never be fixed in legacy branches (470.xx, 390.xx).

I think they will, the same as they adapt the legacy branches to work on older kernels. For sure not in a timely manner :/

jp7677, 2022-05-25:
Kernel 5.17.11, which should contain the workaround based on the kernel-ark repo, arrived in bodhi https://bodhi.fedoraproject.org/updates/FEDORA-2022-8095b23575

stevestorey, 2022-05-26:
And it certainly works well for me, when setting the kernel parameter

scaronni, 2022-05-26:
Will text on monday (it's holiday here) what happens on the optimus laptops when booted with the kernel parameters.

vpxyz, 2022-05-30:
Hi, 
same problem with fedora 35 with kernel 5.17.11. With kernel 5.17.9 everything is fine.
Ciao

stevestorey, 2022-05-30:
The problem isn't with the ark patch - it's that when the resulting changes were picked up and applied to the actual src rpm package, the simpledrm changes weren't reverted - https://src.fedoraproject.org/rpms/kernel/c/8c54d636134215540289a3f8991feb3382ea59f3?branch=f34 - reverting the ark patch won't actually fix the problem

xvitaly, 2022-05-30:
> The problem isn't with the ark patch - it's that when the resulting changes were picked up and applied to the actual src rpm package, the simpledrm changes weren't reverted

`CONFIG_DRM_SIMPLEDRM=y` should be reverted.

stevestorey, 2022-05-30:
Yep, agreed. It's not related to this ticket, or the ARK patch that resolved it though, so can be left to be tracked in the RHBZ's you've opened for them

vpxyz, 2022-05-31:
Ciao, 
today, I don't know why, it works again. I just added "nvidia-drm.modeset=1" to the kernel arguments again.
kernel: 5.17.11-300.fc36.x86_64 driver: nvidia-driver-515.43.04-2.fc36.x86_64

jp7677, 2022-06-01:
The updated kernel works very well here as soon as `nvidia-drm.modeset=1`. That said, exactly this options gets removed when the `nvidia-kmod-common` package is updated (https://github.com/negativo17/nvidia-kmod-common/blob/master/nvidia-kmod-common.spec#L11). Logging in blindly and applying that option again after every update works, but not sure if newcomers will be able to solve this...

scaronni, 2022-06-09:
Ok, I will remove it and re-add it once the Nvidia driver starts supporting `simpledrm`.

GloriousEggroll, 2022-08-21:
I did a fresh clean + install today on two different systems and could not get these drivers to work until `initcall_blacklist=simpledrm_platform_driver_init` was added to the kernel boot options.

[14 more replies]

closed 2023-03-08 by scaronni