negativo17/nvidia-driver #131: Wayland Vulkan swapchain creation fails

flibitijibibo, 2022-05-19:
First, thanks so much for fixing #128! This did fix a number of things right away... but I've now come across another interesting Wayland Vulkan bug, and unlike the other one I'm not sure what's actually happening this time.

The short version: When you try to create a swapchain with a Wayland Vulkan surface, it will fail with `VK_ERROR_INITIALIZATION_FAILED`, with no real indication as to what that actually means. The validation layer is no help either! Just to be sure, I checked an installation with the blob installer and it runs okay. Problem is, unlike the symlink, there doesn't seem to be a clear issue with the Fedora packages at all! I even went and did crazy things like `ls /usr/lib64 > [official|negativo].txt` and diffed the two to be sure we didn't have another obscure file that needed to be there, and no luck.

Based on `LD_DEBUG=libs` and some `strace`ing I think we're now looking at something much weirder - possibly a version sync issue with something like the EGL/GBM libraries??? I can do whatever digging you need on the lab boxes here; the test I'm using is vkcube-wayland:

```
dnf builddep vulkan-tools
git clone https://github.com/KhronosGroup/Vulkan-Tools -b sdk-1.3.204
mkdir Vulkan-Tools/flibitBuild
cd Vulkan-Tools/flibitBuild
cmake ..
make vkcube-wayland
./cube/vkcube-wayland
```

It'll currently crash on an assertion, which is their error check for vkCreateSwapchainKHR.

scaronni, 2022-05-21:
Regarding version mismatches, it's not clear which version of the supporting libraries Nvidia ship in the tarballs when releasing a new driver. My guess is that they just pick the latest commit. The "supporting" libraries are these:

https://github.com/NVIDIA/egl-gbm
https://github.com/NVIDIA/egl-wayland

For egl-gbm we're aligned with the latest release: https://github.com/NVIDIA/egl-gbm/commits/master
For egl-wayland there are 4 extra commits: https://github.com/NVIDIA/egl-wayland/commits/master

I assume you are on Fedora 36? Everything else is up to date. Will try the test you pasted above and let you know.

And last but not least the usual warning: https://forums.developer.nvidia.com/t/wayland-information-for-r515-beta-release/214275

leigh123linux, 2022-05-21:
> Regarding version mismatches, it's not clear which version of the supporting libraries Nvidia ship in the tarballs when releasing a new driver. My guess is that they just pick the latest commit. The [...]

The f36 package is only one commit behind nvidia master.

https://src.fedoraproject.org/rpms/egl-wayland/blob/rawhide/f/egl-wayland.spec#_11

flibitijibibo, 2022-05-21:
Can confirm this is Fedora 36. I can also narrow it down and say the latest egl-wayland commit won't be it, that was something David found while looking at some specific threaded cases for SDL.

scaronni, 2022-05-22:
Then I guess your only option is posting a but in the Nvidia forum.

flibitijibibo, 2022-05-25:
https://forums.developer.nvidia.com/t/negativo17-wayland-vkcreateswapchainkhr-returns-vk-error-initialization-failed/215593

scaronni, 2022-06-01:
Seems to be fixed now? https://forums.developer.nvidia.com/t/linux-solaris-and-freebsd-driver-515-48-07-production-branch-release/216112

I will package 515.48.07 in a moment, it will be online in a few hours.

flibitijibibo, 2022-06-01:
Will check this on the lab boxes today - worst case the debug extension will tell us what's going on.

flibitijibibo, 2022-06-08:
Looks like I was on the right track - from Erik Kurzinger:

>There is an ABI incompatibility between the 515 driver and previous versions of our egl-wayland library which might be the cause of this. Updating to the latest version, 1.1.10, should resolve the issue if that is the case.
>
>Going forward this should never happen again as we’ve eliminated the driver <=> egl-wayland ABI which the Vulkan WSI depended on. However, doing so required one final breaking change, unfortunately.

scaronni, 2022-06-09:
I've pushed a snapshot of 1.1.10 (it seems to be final, but the release has not been tagged). Please let me know how it goes.

Thanks.

jp7677, 2022-06-09:
Running `vkcube-wayland` no longer crashes for me, but fails now with
```
./vkcube-wayland: symbol lookup error: /lib64/libnvidia-vulkan-producer.so: undefined symbol: wlEglInitializeSurfaceExport
```

leigh123linux, 2022-06-09:
> Running `vkcube-wayland` no longer crashes for me, but fails now with
> 
> ```
> ./vkcube-wayland: symbol lookup error: /lib64/libnvidia-vulkan-producer.so: undefined symbol: wlEglInitializeSurfaceExport
> ```

1.1.10 has removed the symbol

https://github.com/NVIDIA/egl-wayland/commit/fee26e9cbd0890bab890f4534d7e92e78d8d0d37

scaronni, 2022-06-10:
Sigh

flibitijibibo, 2022-06-10:
Sigh x2, can reproduce locally.

(As a heads up I may be absent for the next week or so while I get TMNT out the door...)

leigh123linux, 2022-06-10:
It should be fixed in the next driver release.

https://github.com/NVIDIA/egl-wayland/commit/fee26e9cbd0890bab890f4534d7e92e78d8d0d37#commitcomment-75822177

jp7677, 2022-06-28:
A new driver version 515.57 has been released https://www.nvidia.com/download/driverResults.aspx/190422/
Let’s see what this version brings :)

PS: there are also a few new commit for egl-Wayland: https://github.com/NVIDIA/egl-wayland/commits/master as a response to https://forums.developer.nvidia.com/t/properties-and-filters-windows-make-obs-hang-on-wayland-when-closed/213009/12

scaronni, 2022-06-29:
Driver 515.57 is being pushed to the repositories now along with a new snapshot of `egl-wayland`.

flibitijibibo, 2022-06-29:
That's probably accurate - it probably doesn't use decorations and it's using a swapchain with alpha. (Hopefully it's not set to opaque, otherwise I have to go look at my _other_ NV report again...)

Will locally test later today!

lolsniperftw, 2022-06-29:
> A new driver version 515.57 has been released https://www.nvidia.com/download/driverResults.aspx/190422/ Let’s see what this version brings :)
>
> PS: there are also a few new commit for egl-Wayland: [...]

I initially reported the obs issue and there is a weird segfault happening here with their fix:

https://forums.developer.nvidia.com/t/properties-and-filters-windows-make-obs-hang-on-wayland-when-closed/213009/16?u=oranhero

flibitijibibo, 2022-06-29:
Yayyy it works

[image]

Interestingly EGL has opaque presentation (per their recent commits) but Vulkan does not... oh well, back to this issue for that: https://forums.developer.nvidia.com/t/wayland-vulkan-ignores-vkcompositealphaflagskhr/202227

Thanks again for all the help!

closed 2022-06-29 by flibitijibibo

darkestmedium, 2024-08-20:
[image]

I think you can close it now. 550 works. :>