negativo17/nvidia-driver #132: 515.48.07 is a stable release

RyzenDew, 2022-06-01:
https://www.nvidia.com/Download/driverResults.aspx/189809/en-us please update to the newest

Published the source code to a variant of the NVIDIA Linux kernel modules dual-licensed as MIT/GPLv2. The source is available here:
https://github.com/NVIDIA/open-gpu-kernel-modules
and will be updated each driver release. Please see the "Open Linux Kernel Modules" chapter in the README for details.
Added support for the VK_EXT_external_memory_dma_buf and VK_EXT_image_drm_format_modifier Vulkan extensions. To use this functionality, the nvidia-drm kernel module must be loaded with DRM KMS mode setting enabled. See the DRM KMS section of the README for guidance on enabling mode setting.
Changed nvidia-suspend.service, nvidia-resume.service, and nvidia-hibernate.service to use WantedBy= rather than RequiredBy= dependencies for systemd-suspend.service and systemd-hibernate.service. This avoids a problem where suspend or hibernate fails if the NVIDIA driver is uninstalled without disabling these services first.
See https://github.com/systemd/systemd/issues/21991
If these services were manually enabled, it may be necessary to update their dependencies by running
sudo systemctl reenable nvidia-suspend.service nvidia-resume.service nvidia-hibernate.service
Interlaced modes are now disabled when active stereo is enabled.
NVIDIA X Server Settings will now display the quit confirmation dialog automatically if only there are pending changes that need to be manually saved. The corresponding configuration option to control the appearance of the quit dialog was thus also removed.
Removed the warning message about mismatches between the compiler used to build the Linux kernel and the compiler used to build the NVIDIA kernel modules from nvidia-installer. Modern compilers are less likely to cause problems when this type of mismatch occurs, and it has become common in many distributions to build the Linux kernel with a different compiler than the default system compiler.
Updated nvidia-installer to skip test-loading the kernel modules on systems where no supported NVIDIA GPUs are detected.
Updated nvidia-installer to avoid a race condition which could cause the kernel module test load to fail due to udev automatically loading kernel modules left over from an existing NVIDIA driver installation. This failure resulted in an installation error message "Kernel module load error: File exists".
Updated the RTD3 Video Memory Utilization Threshold (NVreg_DynamicPowerManagementVideoMemoryThreshold) maximum value from 200 MB to 1024 MB.
Improved performance of GLX and Vulkan applications running in gamescope.
Added a "kernelopen" feature tag to the supported-gpus.json file, to indicate which GPUs are compatible with open-gpu-kernel-modules.
Improved Vulkan swapchain creation failure reporting. Applications can use the VK_EXT_debug_utils extension to receive additional information when an error was encountered in vkCreateSwapchainKHR().
Added a new configuration option for NVIDIA NGX to allow disabling the DSO signature check. See the "NGX" chapter of the README for more information.
Fixed an issue where HDMI audio output was not working in some cases, especially with high display refresh rates (120Hz, 100Hz, etc.) using Fixed Rate Link (FRL) transmission mode.

scaronni, 2022-06-01:
Hi, yes, I've seen it, please give me a moment to update it :)

Packages will be online in a few hours.

RyzenDew, 2022-06-01:
I don’t see the repo updated for it yet .. how is it completed confused ..