negativo17/nvidia-driver #134: No nvidia-settings for Fedora 37 and the latest driver version (525.60.11)

mwawra, 2022-12-01:
Hi,
during the update I just recognized that there is no package for nvidia-settings for the latest driver version (525.60.11) in the repo for Fedora 37.

msmafra, 2022-12-01:
I was having some conflicts with RPMFusion, disabled all RPMFusion repos, but there are miss matching versions:
```
❯ doas dnf install akmod-nvidia nvidia-driver nvidia-modprobe nvidia-driver-libs.i686 nvidia-driver-NVML.{x86_64,i686} nvidia-libXNVCtrl nvidia-driver-NvFBCOpenGL.{x86_64,i686} nvidia-xconfig nvidia-settings nvidia-driver-cuda-\*.{x86_64,i686} cuda-devel nvidia-driver-cuda
Last metadata expiration check: 0:03:35 ago on Thu Dec  1 16:48:14 2022.
Dependencies resolved.

 Problem: cannot install both nvidia-driver-3:525.60.11-1.fc37.x86_64 and nvidia-driver-3:520.56.06-1.fc37.x86_64
  - package nvidia-settings-3:520.56.06-1.fc37.x86_64 requires nvidia-driver(x86-64) = 3:520.56.06, but none of the providers can be installed
  - cannot install the best candidate for the job
  - conflicting requests
=====================================================================================================================================================================================================================
 Package                                                     Architecture                             Version                                              Repository                                           Size
=====================================================================================================================================================================================================================
Installing:
 akmod-nvidia                                                x86_64                                   3:520.56.06-1.fc37                                   negativo17-nvidia                                    41 M
 cuda-devel                                                  x86_64                                   1:11.8.86-1.fc37                                     negativo17-nvidia                                   9.0 k
 nvidia-driver-NVML                                          i686                                     3:525.60.11-1.fc37                                   negativo17-nvidia                                   469 k
 nvidia-driver-NvFBCOpenGL                                   x86_64                                   3:520.56.06-1.fc37                                   negativo17-nvidia                                    55 k
 nvidia-driver-NvFBCOpenGL                                   i686                                     3:525.60.11-1.fc37                                   negativo17-nvidia                                    60 k
 nvidia-driver-cuda                                          x86_64                                   3:520.56.06-1.fc37                                   negativo17-nvidia                                   1.7 M
 nvidia-driver-libs                                          i686                                     3:525.60.11-1.fc37                                   negativo17-nvidia                                    29 M
 nvidia-libXNVCtrl                                           x86_64                                   3:520.56.06-1.fc37                                   negativo17-nvidia                                    25 k
 nvidia-modprobe                                             x86_64                                   3:525.60.11-1.fc37                                   negativo17-nvidia                                    29 k
 nvidia-settings                                             x86_64                                   3:520.56.06-1.fc37                                   negativo17-nvidia                                   813 k
 nvidia-xconfig                                              x86_64                                   3:525.60.11-1.fc37                                   negativo17-nvidia                                    95 k
Installing dependencies:
 cuda                                                        x86_64                                   1:11.8.86-1.fc37                                     negativo17-nvidia                                    82 k
 cuda-cudart                                                 x86_64                                   1:11.8.89-1.fc37                                     negativo17-nvidia                                   177 k
 cuda-cudart-devel                                           x86_64                                   1:11.8.89-1.fc37                                     negativo17-nvidia                                   627 k
 cuda-cuobjdump                                              x86_64                                   1:11.8.86-1.fc37                                     negativo17-nvidia                                   184 k
 cuda-cupti                                                  x86_64                                   1:11.8.87-1.fc37                                     negativo17-nvidia                                   2.0 M
 cuda-cupti-devel                                            x86_64                                   1:11.8.87-1.fc37                                     negativo17-nvidia                                   8.9 M
 cuda-cuxxfilt-devel                                         x86_64                                   1:11.8.86-1.fc37                                     negativo17-nvidia                                   210 k
[... cut ...]
 nvidia-driver                                               x86_64                                   3:525.60.11-1.fc37                                   negativo17-nvidia                                   2.6 M

Transaction Summary
=====================================================================================================================================================================================================================
Install  51 Packages
Skip      1 Package

Total download size: 1.1 G
Installed size: 3.3 G
Is this ok [Y/n]: 

```

zeroepoch, 2022-12-02:
I believe this will be fixed by https://github.com/negativo17/nvidia-settings/commit/da1649ff5f51088b9b63d7b0799f3d2e8c488f8a

scaronni, 2022-12-04:
Argh, sorry, small issue that prevented the package being pushed. I was away for the weekend, coming online in a few minutes.

kmare, 2022-12-07:
> Argh, sorry, small issue that prevented the package being pushed. I was away for the weekend, coming online in a few minutes.

@scaronni when do you think it's going to be pushed? At the moment it's not upgrading to the last 525.
Thank you for your dedication btw!

scaronni, 2022-12-07:
Argh, I synced only the multimedia repository but not the nvidia one: https://negativo17.org/repos/multimedia/fedora-37/x86_64/

Syncing now, sorry for the trouble!

mwawra, 2022-12-08:
Thanks for solving that! 👍

kmare, 2022-12-08:
No worries, thank you! It worked great!