negativo17/nvidia-driver #135: Conflict with official cuda nvidia-driver-cuda package

GloriousEggroll, 2022-12-05:
So, first a small understanding of how I got here:

(1) The cuda + cuda-devel packages provided in the Negativo17 repo do not work with Davinci Resolve Studio (the paid version). If you install them, H264/H265 encode options are not enabled in DRS.

Which lead me to:

(2) The cuda packages provided by the official nvidia repo work properly with DRS, but they cause SDDM to crash (for KDE users). Lightdm also does not work. Furthermore nvidia's offical repo disables the 'nvidia-driver' package module, so you can't mix it's cuda packages with other nvidia packages unless you explicity install the nvidia-driver packages you want first, then install cuda. They also provide an older version of the current drivers so they are outdated.

This then lead me to:

(3) Since the Negativo17 repo docs state:

https://negativo17.org/nvidia-driver/#CUDA_support
```
Complete packaged CUDA stack has been added for all supported distributions, all the packages provide/require/obsolete the relevant packages in the [Nvidia CUDA repository](http://developer.download.nvidia.com/compute/cuda/repos/); so you can enable this repository along with the official Nvidia CUDA one (x86_64 systems only
```

But said cuda packages did not work properly, I thought "hey, maybe I can use the nvidia drivers from Negativo17, and the cuda drivers/packages from the official Nvidia repo"

That's when I hit a speedbump:

/usr/bin/nvidia-powerd is provided by the nvidia-driver package in Negativo17, but it's provided by the nvidia-driver-cuda package in Nvidia's official repo, which means you cant directly upgrade/downgrade between the two. You have to completely remove all the nvidia official packages, then install just the specific negativo17 packages for nvidia drivers, then install the cuda drivers from nv official.

So... basically the TLDR is:
The powerd binary should be moved so interchangeability is possible, and a second bug (opened here: https://github.com/negativo17/cuda/issues/29) should be added to fix the Negativo17 cuda packages not providing H264/H265 Nvidia encode options in DRS.

scaronni, 2022-12-20:
So this is a bug on the Nvidia side, `nvidia-powerd` is required for Max-Q laptops which have the "SBIOS support for Dynamic Boost":

https://download.nvidia.com/XFree86/Linux-x86_64/525.60.11/README/dynamicboost.html

It has nothing to do with CUDA. Also, some other differences:

- File locations:
```
$ rpm -qpl nvidia-driver-cuda-525.60.13-1.fc36.x86_64.rpm | grep powerd
/usr/bin/nvidia-powerd
$ rpm -qpl nvidia-driver-525.60.13-1.fc36.x86_64.rpm | grep powerd
/usr/lib/systemd/system/nvidia-powerd.service
```
- I enable the various systemd unit (`nvidia-hibernate`, `nvidia-resume`, `nvidia-suspend` and `nvidia-powerd`) from a `systemd` preset file in the main package, along with the `systemd` units. They don't enable anything by default.
- I have the firmware files inside `nvidia-kmod-common`, they have them in the main package.

Besides that, everything should update without issues if you install the same set of packages as all the files are there. I'm thinking if I should just simplify all packaging and just specify that the repositories can't be used together.

GloriousEggroll, 2022-12-31:
> So this is a bug on the Nvidia side, `nvidia-powerd` is required for Max-Q laptops which have the "SBIOS support for Dynamic Boost": [...]

since the underlying cuda issue is resolved i would agree that just going ahead and noting that nvidia's cuda repo packages are not compatible with these, it would most definitely save some headache. thanks again for getting this stuff sorted out, it's really appreciated

scaronni, 2024-04-27:
The official CUDA drivers ~560+ will be in sync with my packages, both in terms of dependencies as well as features. I will make sure both will be compatible (I work for Nvidia now).

closed 2024-04-27 by scaronni