negativo17/nvidia-driver #136: Conditional start for nvidia-powerd

scaronni, 2022-12-20:
If someone happens to run Nvidia drivers on a Max-Q laptop, I would like to know if there is a way to detect "SBIOS support for Dynamic Boost" by using `dmidecode`.

With that information, we could add a check command to the `nvidia-powerd` `systemd` unit to only start in case the laptop is actually a Max-Q laptop.

GloriousEggroll, 2022-12-31:
not sure about dmidecode but per the docs there is a checkable option from nvidia-settings:

https://download.nvidia.com/XFree86/Linux-x86_64/510.60.02/README/dynamicboost.html

```

SBIOS support for Dynamic Boost

Run the following command to check if the SBIOS supports Dynamic Boost:

nvidia-settings -q DynamicBoostSupport
```

scaronni, 2023-01-02:
`nvidia-settings` is not required to run the driver, it's optionally installable. I would like to get it from the appropriate "source", e.g. the DMI table.

GloriousEggroll, 2023-12-16:
I've proposed something here: https://github.com/negativo17/nvidia-driver/pull/148  that should work. It's not using dmi table but it seems good enough here at least to where users arent questioning why it's failing. We're using it on Nobara.

scaronni, 2023-12-18:
Hi, I like the idea, I would take a similar approach, but using a drop in customization for the `systemd` unit (ex. `/usr/lib/systemd/system/nvidia-powerd.service.d/something.conf`). Why?

For EL9 / Fedora (`systemd` 243+) the `ExecCondition=` directive can be used (https://www.freedesktop.org/software/systemd/man/latest/systemd.service.html) in place of editing `ExecStart`, and on EL7/EL8 we can just override the `ExecStart`.

Thoughts?

scaronni, 2023-12-18:
I'll come up with something based on your MR or an update from, whichever comes first, thanks for the input!

I'm planning to make some tests to update `nvidia-kmod-common` for the `sdboot` case.

scaronni, 2023-12-18:
Actually... it just came to my mind now. Why not just ignore the return code? Then it's just listed as being turned off:

```
# systemctl status nvidia-powerd.service 
○ nvidia-powerd.service - nvidia-powerd service
     Loaded: loaded (/usr/lib/systemd/system/nvidia-powerd.service; enabled; preset: enabled)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
             /usr/lib/systemd/system/nvidia-powerd.service.d
             └─condition.conf
     Active: inactive (dead) since Mon 2023-12-18 13:16:10 CET; 5s ago
   Duration: 22ms
    Process: 2172487 ExecStart=/usr/bin/nvidia-powerd (code=exited, status=1/FAILURE)
   Main PID: 2172487 (code=exited, status=1/FAILURE)
        CPU: 15ms

Dec 18 13:16:10 example.com systemd[1]: Started nvidia-powerd.service - nvidia-powerd service.
Dec 18 13:16:10 example.com /usr/bin/nvidia-powerd[2172487]: nvidia-powerd version:1.0(build 1)
Dec 18 13:16:10 example.com /usr/bin/nvidia-powerd[2172487]: No matching GPU found
Dec 18 13:16:10 example.com /usr/bin/nvidia-powerd[2172487]: Failed to initialize RM Client
Dec 18 13:16:10 example.com systemd[1]: nvidia-powerd.service: Deactivated successfully.
# systemctl --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.
```

It just needs a dash in front of the binary name:
```
ExecStart=-/usr/bin/nvidia-powerd
```

GloriousEggroll, 2023-12-19:
that could work as well, feel free to drop whatever you have here, i can test and report back. we're getting ready to release n39 so i wanted something in place as this is a bit of a pain point for new users

closed 2023-12-21 by scaronni

scaronni, 2023-12-21:
As mentioned a few comments above, unrelated but might be relevant for you: https://github.com/negativo17/nvidia-kmod-common/issues/10#issuecomment-1865876167