negativo17/nvidia-driver #137: Missing dependencies for mesa-vdpau-drivers

Dapid, 2023-02-14:
I am getting this message after updating to Fedora 37:

```
 Problem: cannot install the best update candidate for package mesa-vdpau-drivers-22.3.4-1.fc37.x86_64
  - nothing provides mesa-filesystem(x86-64) = 1:22.3.4-1.fc37 needed by mesa-vdpau-drivers-1:22.3.4-1.fc37.x86_64
```

Comparing `mesa-vdpau-drivers` between the updates repo and this, one, the versions match, except for the epoch.

```
Name         : mesa-vdpau-drivers
Version      : 22.3.4
Release      : 1.fc37
Architecture : x86_64
Size         : 12 M
Source       : mesa-22.3.4-1.fc37.src.rpm
Repository   : @System

Name         : mesa-vdpau-drivers
Epoch        : 1              ##### <--- Difference
Version      : 22.3.4
Release      : 1.fc37
Architecture : x86_64
Size         : 3.5 M
Source       : mesa-22.3.4-1.fc37.src.rpm
Repository   : fedora-nvidia-negativo17
```

`mesa-filesystem` is only present in the `@updates` repository, but it seems it is expecting a corresponding "epoched" version here?

It isn't a big problem for me, just a bit of extra output when doing `dnf update`, but I thought to report it just in case it is a symptom of something else.

scaronni, 2023-02-25:
Sorry for the late reply, but the package is there and I'm not having the issue:

```
$ rpm -qpi fedora-37/x86_64/mesa-filesystem-22.3.4-1.fc37.x86_64.rpm
Name        : mesa-filesystem
Epoch       : 1
Version     : 22.3.4
Release     : 1.fc37
Architecture: x86_64
Install Date: (not installed)
Group       : Unspecified
Size        : 3682
License     : MIT
Signature   : DSA/SHA1, Tue 07 Feb 2023 06:28:03 PM CET, Key ID 14386362f90c0e97
Source RPM  : mesa-22.3.4-1.fc37.src.rpm
Build Date  : Tue 07 Feb 2023 05:20:56 PM CET
Build Host  : workstation.localdomain
URL         : http://www.mesa3d.org
Summary     : Mesa driver filesystem
Description :
Mesa driver filesystem.
```

scaronni, 2023-02-25:
Do you still have the issue?