negativo17/nvidia-driver #138: 530.30.02  vs 530.41.03

msmafra, 2023-03-28:
Hello there!

I'm on Fedora 38 Beta on my laptop and while I was using NVIDIA drivers 530.30.02 (Beta) I had the ability to use GreenWithEnvy power limit feature. Not just to change power limits but to monitor, after the update to 530.41.03 is partially not working. Now I can't see Min, Max, Limit, Default or Enforced only see the Draw and can't set a Limit anymore. I checked the .conf files and they have not been replaced or modified. As I'm using F38 Beta, I'm not sure if it's a driver bug, DNF or DNF5 or a system bug (Gnome, kernel etc). 
Anyone else with a laptop with hybrid graphics having those issues?

`doas dnf5 install akmod-nvidia nvidia-driver nvidia-modprobe nvidia-driver-libs.{i686,x86_64} nvidia-driver-NVML.{x86_64,i686} nvidia-libXNVCtrl nvidia-xconfig nvidia-settings nvidia-driver-NvFBCOpenGL.{i686,x86_64} nvidia-driver-cuda nvidia-driver-cuda-libs.{x86_64,i686}`
[image]
[image]

[image]

scaronni, 2023-03-30:
Hi, first of all you would need to check with the upstream GreenWithEnvy project. According to this issue, the developer has switched to AMD: https://gitlab.com/leinardi/gwe/-/issues/195

Second, the configuration files you're showing above are not coming from my packages, some of them have completely different content. These are the files I ship, compare them if you want:
```
$ rpm -ql nvidia-kmod-common nvidia-driver | grep conf$
/etc/modprobe.d/nvidia-modeset.conf
/usr/lib/dracut/dracut.conf.d/99-nvidia.conf
/usr/lib/modprobe.d/nvidia.conf
/etc/X11/xorg.conf.d/10-nvidia.conf
/usr/share/dbus-1/system.d/nvidia-dbus.conf
```
Closing this down.

closed 2023-03-30 by scaronni

msmafra, 2023-03-30:
Hello there!
Sorry, I did not explain it correctly. Those modification I made on my system. What I meant was that with those same settings, that I modified because of unbearable tearing on 530.30.02 beta on my system, and the same version of GWE, I was able to use the power section. 

Using those dracut settings, helps booting my dual monitor setup (laptop screen + external monitor) making Plymouth's LUKS prompt or the verbose boot process  be shown on both monitors when NVIDIA drivers are installed, which is not the default. I'm imagining that probably NVIDIA changed something before the final release as 530.41.03.

I'm aware of Leinardi's leaving the project and hoping someone keep it running.

Thanks.