negativo17/nvidia-driver #139: Weird corruption/misrendering in games

dvdmuckle, 2023-03-31:
I've been having issues with the latest driver (530.41) in some games running through Proton. I've tested this with the same driver version from rpmfusion and those seem to work fine, so I'm opening an issue here. I'm not sure how that could make a difference but it's the only change I've made that fixed the issue. This is with Fedora 37 on a machine with both an RTX 3070ti and a mobile 1650. Example of the bug attached
[image]

dvdmuckle, 2023-04-01:
My mistake, this looks to happen with the RPMFusion version too. I guess this must be a bug with this driver...? Or perhaps to do with shaders. Either way, closing this issue.

closed 2023-04-01 by dvdmuckle

leigh123linux, 2023-04-01:
Maybe it's a steam issue.

https://bugzilla.rpmfusion.org/show_bug.cgi?id=6615

dvdmuckle, 2023-04-01:
Hm, that sounds like it might be the case. I'm hearing from some folks that the 530 driver itself might not be in great shape, but I can't find any specific issue or bug filed.