negativo17/nvidia-driver #143: Problems when RPMfusion goes out of sync

Dapid, 2023-08-24:
When RPMfusion publishes a newer driver, DNF will prefer it and make a mess, installing a mix of both. For example, on my system right now, it wants to install `akmod-nvidia`, `nvidia-modprobe`, and `nvidia-xconfig` while leaving the driver from Negativo.

I suspect the right way to solve it is to make the dependencies more explicit, but in the meantime, I added the following line to `/etc/yum.repos.d/rpmfusion-nonfree-updates.repo` and `/etc/yum.repos.d/rpmfusion-nonfree.repo`, so it only pulls from Negativo17:

```
exclude=akmod-nvidia,nvidia-modprobe,nvidia-xconfig,nvidia-persistenced,nvidia-settings,nvidia-driver,nvidia-libXNVCtrl,nvidia-driver-libs
```

scaronni, 2023-08-24:
I think if you want to use RPMFusion you should use the Nvidia driver packaged there. I'm not testing for compatibility.

Dapid, 2023-08-24:
To be clear, I am using the driver from Negativo17, and RPMFusion for other, unrelated, packages. That has been my setup for years.

Is that unsupported?

scaronni, 2023-08-24:
Yes it's unsupported.

In the past, someone reported something similar, so I bumped the Epoch of the packages to be used together, and in retaliation RPMFusion bumped it again for whatever reason.

The packages with different names conflict, but similarly named ones are updated to the newest, wherever it comes from.

If you want to use both, I guess your repository setup is probably the best you can get.

closed 2023-08-24 by scaronni

leigh123linux, 2023-08-24:
> Yes it's unsupported.
> 
> In the past, someone reported something similar, so I bumped the Epoch of the packages to be used together, and in retaliation RPMFusion bumped it again for whatever reason.
> 
I bumped the epoch because nvida cuda bumped theirs.

https://github.com/rpmfusion/xorg-x11-drv-nvidia/commit/de92ef6e74aa29ca8648622692670d3e893b9061

leigh123linux, 2023-08-24:
And before that I rolled back the nvidia version which required an epoch.

https://github.com/rpmfusion/xorg-x11-drv-nvidia/commit/f6ac3668c886651dc4242bb25a021e1b76e4b9b0

leigh123linux, 2023-08-24:
I see no malice to the epoch bumps.

https://github.com/rpmfusion/xorg-x11-drv-nvidia/commit/bd7699d1ccaaa5a70703f407b85790e202b1c706

scaronni, 2023-08-28:
To be honest by now I can't remember, too much time has passwed. It was when Hans and Red Hat was trying to merge stuff together, maybe it was just a coincidence :peace_symbol: