negativo17/nvidia-driver #144: New driver release : 545.29.02

DimPaDev, 2023-10-31:
I just wanted to kindly ask if you could spare a bit of your time to update the Nvidia drivers in the repository. The new drivers with version number 545.29.02 are out, and they seem to be a promising update!

@scaronni 

 [https://www.phoronix.com/news/NVIDIA-545.29.02-Linux-Driver](url)

scaronni, 2023-10-31:
Yes indeed it's a good one. Also the previous beta was working fine for my setups.

Building it now, will be online in a few hours.

closed 2023-10-31 by scaronni

leigh123linux, 2023-10-31:
This release is no better than beta quality, I'm skipping for < f40 bue to broken vulkan under wayland and this issue.

https://forums.developer.nvidia.com/t/cs2-stuck-on-valve-logo-on-startup-545-beta-branch/269778/19

scaronni, 2023-11-01:
Bummer. Never played counterstrike though.

leigh123linux, 2023-11-01:
> Bummer. Never played counterstrike though.

I doubt the issues are limited to c-s.
545.xx caused basic vulkan apps like vkcube to crash under wayland.

Xerkus, 2023-11-01:
Apparently that release along with 535.129.03, 525.147.05 and 470.223.02 are security releases: https://nvidia.custhelp.com/app/answers/detail/a_id/5491

Xerkus, 2023-11-01:
545 seems to be busted.

After update to 545 from 535 steam immediately segfaults on launch with nothing else changed.  I made sure I got no lingering libraries or kernel modules from 535.
It is launching under xwayland on nvidia with prime offload. Launching on iGPU has no issues. 

```
Packages Altered:
    Upgrade  dkms-nvidia-3:545.29.02-1.fc38.x86_64              @fedora-nvidia
    Upgraded dkms-nvidia-3:535.113.01-1.fc38.x86_64             @@System
    Upgrade  nvidia-driver-3:545.29.02-1.fc38.x86_64            @fedora-nvidia
    Upgraded nvidia-driver-3:535.113.01-1.fc38.x86_64           @@System
    Upgrade  nvidia-driver-NVML-3:545.29.02-1.fc38.x86_64       @fedora-nvidia
    Upgraded nvidia-driver-NVML-3:535.113.01-1.fc38.x86_64      @@System
    Upgrade  nvidia-driver-cuda-3:545.29.02-1.fc38.x86_64       @fedora-nvidia
    Upgraded nvidia-driver-cuda-3:535.113.01-1.fc38.x86_64      @@System
    Upgrade  nvidia-driver-cuda-libs-3:545.29.02-1.fc38.i686    @fedora-nvidia
    Upgraded nvidia-driver-cuda-libs-3:535.113.01-1.fc38.i686   @@System
    Upgrade  nvidia-driver-cuda-libs-3:545.29.02-1.fc38.x86_64  @fedora-nvidia
    Upgraded nvidia-driver-cuda-libs-3:535.113.01-1.fc38.x86_64 @@System
    Upgrade  nvidia-driver-libs-3:545.29.02-1.fc38.i686         @fedora-nvidia
    Upgraded nvidia-driver-libs-3:535.113.01-1.fc38.i686        @@System
    Upgrade  nvidia-driver-libs-3:545.29.02-1.fc38.x86_64       @fedora-nvidia
    Upgraded nvidia-driver-libs-3:535.113.01-1.fc38.x86_64      @@System
    Upgrade  nvidia-kmod-common-3:545.29.02-1.fc38.noarch       @fedora-nvidia
    Upgraded nvidia-kmod-common-3:535.113.01-1.fc38.noarch      @@System
    Upgrade  nvidia-persistenced-3:545.29.02-1.fc38.x86_64      @fedora-nvidia
    Upgraded nvidia-persistenced-3:535.113.01-1.fc38.x86_64     @@System
```

```
Stack trace of thread 9475:
#0  0x00000000f7c33453 strchr (libc.so.6 + 0xa3453)
#1  0x00000000f7c3625b strstr (libc.so.6 + 0xa625b)
#2  0x00000000eb6d1682 n/a (/home/Xerkus/.local/share/Steam/ubuntu12_32/vgui2_s.so + 0x17682)
#3  0x00000000eb714aec n/a (/home/Xerkus/.local/share/Steam/ubuntu12_32/vgui2_s.so + 0x5aaec)
#4  0x00000000eb723e2f n/a (/home/Xerkus/.local/share/Steam/ubuntu12_32/vgui2_s.so + 0x69e2f)
#5  0x00000000ef59442a n/a (/home/Xerkus/.local/share/Steam/ubuntu12_32/steamui.so + 0xd9442a)
#6  0x00000000eec13fbc n/a (/home/Xerkus/.local/share/Steam/ubuntu12_32/steamui.so + 0x413fbc)
#7  0x00000000565cbbe9 n/a (/home/Xerkus/.local/share/Steam/ubuntu12_32/steam + 0x4abe9)
#8  0x00000000565ccafe n/a (/home/Xerkus/.local/share/Steam/ubuntu12_32/steam + 0x4bafe)
#9  0x00000000565b6bb0 n/a (/home/Xerkus/.local/share/Steam/ubuntu12_32/steam + 0x35bb0)
```

Same deal with Stellaris, got segfault:
```
#0  0x00007f2dc23e4f5b n/a (libGLX_nvidia.so.0 + 0x53f5b)
#1  0x00007f2dc2411e3a n/a (libGLX_nvidia.so.0 + 0x80e3a)
#2  0x00007f2dc36c8701 InternalLoseCurrent (libGLX.so.0 + 0x16701)
#3  0x00007f2dc36cd152 CommonMakeCurrent (libGLX.so.0 + 0x1b152)
#4  0x00007f2dbabaa4bd n/a (/home/Xerkus/.paradoxlauncher/launcher-v2.2023.13.1/libGLESv2.so + 0x3aa4bd)
#5  0x00007f2dba9d907e n/a (/home/Xerkus/.paradoxlauncher/launcher-v2.2023.13.1/libGLESv2.so + 0x1d907e)
#6  0x00007f2dba965e10 n/a (/home/Xerkus/.paradoxlauncher/launcher-v2.2023.13.1/libGLESv2.so + 0x165e10)
#7  0x00007f2dba967fde n/a (/home/Xerkus/.paradoxlauncher/launcher-v2.2023.13.1/libGLESv2.so + 0x167fde)
#8  0x000055bf22832905 n/a (/home/Xerkus/.paradoxlauncher/launcher-v2.2023.13.1/Launcher + 0x5cc8905)
#9  0x000055bf2283272d n/a (/home/Xerkus/.paradoxlauncher/launcher-v2.2023.13.1/Launcher + 0x5cc872d)
```

glxgears does not run on nvidia, runs on igpu. eglgears_x11 (took binary from mesa-demos from rawhide) ignores env variables and runs with igpu. eglgears_wayland runs with nvidia but no window appears and runs normally with igpu.

```
$ __GLX_VENDOR_LIBRARY_NAME=nvidia __NV_PRIME_RENDER_OFFLOAD=1 __VK_LAYER_NV_optimus=NVIDIA_only glxgears -info
X Error of failed request:  BadAlloc (insufficient resources for operation)
  Major opcode of failed request:  150 (GLX)
  Minor opcode of failed request:  5 (X_GLXMakeCurrent)
  Serial number of failed request:  0
  Current serial number in output stream:  36
```

No clue what to make of this. Going to downgrade to 535 for now.

kmare, 2023-11-01:
@scaronni 
Thank you for the prompt release!
Unfortunately it seems like something is broken. After upgrading from 535 to 545, I have only LLVM (software) rendering. The module seems to be installed and loaded (lsmod shows nvidia loaded and running). But a simple glxinfo will return LLVM rendering. Also running nvidia-smi returns an error saying that the driver is not loaded/running.
Had to downgrade to get back to "normal".
Any help is highly appreciated!

DimPaDev, 2023-11-01:
yes I have the same problem as @kmare 

Tested on a laptop with prime (intel+nvidia)

Xerkus, 2023-11-02:
@kmare are you using dkms? In my experience nvidia-smi fails when somehow module is compiled for a new just installed kernel using current kernel libraries. I don't think driver not loaded is a problem caused by this release.

Try to wipe compiled module and then build/install it again. Eg `sudo dkms remove nvidia/545.29.02 --all` then  `sudo dkms autoinstall`. You can get exact version with dkms status

kmare, 2023-11-02:
@Xerkus thank you, but unfortunately it didn't work for me. Neither the laptop or the PC. I don't know what it could be, maybe it's the way the drivers are getting signed and not properly loaded? 
To be honest I'm not sure if it's the drivers themselves or the packaging (I'm not too keen in installing directly the officially provided tarball).
I hope scaronni comes up with a fix soon :)

kmare, 2023-11-03:
Fixed!
I just edited the nvidia udev rule (/usr/lib/udev/rules.d/60-nvidia.rules) and replaced "nvidia-frontend" with "nvidiactl" after installing the driver.
Running the drivers smoothly on wayland now :)