negativo17/nvidia-driver #145: Fedora 37 - GTX 1070 - 545.29.02-1 - Xorg Failed to initialize the NVIDIA kernel module.

mvrk69, 2023-11-03:
Hi,

I just upgraded from 535.113.01-1 to 545.29.02-1 and Xorg can't initialize nvidia module.

In attachment the Xorg and kernel logs.

Anyone experienced something like this?

[kernel.log](https://github.com/negativo17/nvidia-driver/files/13252884/kernel.log)
[Xorg.0.log](https://github.com/negativo17/nvidia-driver/files/13252885/Xorg.0.log)

leigh123linux, 2023-11-03:
The udev rules are broken in 545.xx, see https://bugzilla.rpmfusion.org/show_bug.cgi?id=6784

kmare, 2023-11-03:
@leigh123linux thanks for sharing! 
Did you manage to get it to work by any chance?

leigh123linux, 2023-11-03:
GDM was able to log into a xorg session after deleting the nvidia udev rule and installing nvidia-modprobe.

kmare, 2023-11-03:
Actually, I just edited the udev rule (/usr/lib/udev/rules.d/60-nvidia.rules) and replaced "nvidia-frontend" with "nvidiactl" after installing the driver.
Now it seems to be working fine! Thank you @leigh123linux !
@scaronni maybe you could consider applying this fix? Thank you in advance!

scaronni, 2023-11-04:
> Actually, I just edited the udev rule (/usr/lib/udev/rules.d/60-nvidia.rules) and replaced "nvidia-frontend" with "nvidiactl" after installing the driver. Now it seems to be working fine! Thank you @leigh123linux ! @scaronni maybe you could consider applying this fix? Thank you in advance!

Correct, this seems the proper fix: https://github.com/negativo17/nvidia-kmod-common/issues/11

The udev rule is needed when the display needs to be brought up without the DDX driver being loaded (as in the X case). I'll push a fix now on the `nvidia-kmod-common` package.

I'll track it separately in https://github.com/negativo17/nvidia-kmod-common/issues/11

closed 2023-11-04 by scaronni

scaronni, 2023-11-04:
> The udev rule is needed when the display needs to be brought up without the DDX driver being loaded (as in the X case). I'll push a fix now on the `nvidia-kmod-common` package.

Just to add to it, the new framebuffer console does not start until an X session is started if using `nvidia-modprobe` only, it's written in the issue linked above.