negativo17/nvidia-driver #146: Brightness control 545.29 regression

msmafra, 2023-11-11:
Updating to 545.29.2 and not able to control brightness  of my laptop screen I thought it could have something to do with the rules files and my Legion 5i, but I saw [here](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/573) (issue #4364327) that it seems to be also in the NVIDIA proprietary. Not sure if it's for all laptops or specific ones, so I tested with 535.1290 too since it was also released on the same date as 545.29, 525.1470 and 470.2230, and works on 535.129.03.
There is a patch the to apply that seems to solves it, not sure if it's viable to you to apply it.

scaronni, 2023-11-12:
The patch is for the open modules, not the proprietary ones. Considering the open modules are beta quality in 545, maybe it's time to add them, and add the patch as well. I **don't** have hardware to test them.

msmafra, 2023-11-12:
Oh, is the open modules. I was about to get back to 535 I updated to 545 mainly to try finally migrating to Wayland. I don't mind testing it.

scaronni, 2023-11-13:
@msmafra are you using `dkms-nvidia` or `akmod-nvidia` on your system? I have something to test for this.

msmafra, 2023-11-13:
> @msmafra are you using `dkms-nvidia` or `akmod-nvidia` on your system? I have something to test for this.

I'm using akmod-nvidia

scaronni, 2023-11-14:
Can you try the following? I've set the option to build proprietary or open source modules in the drivers (both akmod and DKMS) by changing a variable.

The open source variant of the modules contains your patch for the backlight. Please let me know here how it goes as I currently don't have any GSP based hardware to test.

Instructions:

Add this temporary repository to your system (it's an interim repo I use for rebuilding stuff internally):
```
# curl https://uno.negativo17.org/repos/fedora-nvidia-uno.repo > /etc/yum.repos.d/fedora-nvidia-uno.repo
```
Update your system, you should get at minimum a new `akmod-nvidia` and `nvidia-kmod-common` package:
```
# dnf update
```
Now check the variables and the packages on your system, it should be pretty self explanatory:
```
# modinfo -l nvidia
NVIDIA
# vi /etc/nvidia.conf
# akmods --rebuild
Checking kmods exist for 6.5.11-300.fc39.x86_64            [  OK  ]
Building and installing nvidia-kmod                        [  OK  ]
# modinfo -l nvidia
Dual MIT/GPL
```
Then reboot, you should be online again with the open source drivers and the backlight patch. You can now delete `/etc/yum.repos.d/fedora-nvidia-uno.repo`, I will push the packages to the main repository soon with some instructions on the blog.

msmafra, 2023-11-15:
> Can you try the following? I've set the option to build proprietary or open source modules in the drivers (both akmod and DKMS) by changing a variable.
>
> The open source variant of the modules [...]

SOrry. I'll try tonight. Did not check earlier. Heatwave here... just now can barely think :). I'll let you know.

msmafra, 2023-11-15:
I had already uninstalled it. Installed it again with the new repo enabled. Tested with Fedora 39's kernel 6.5.11-300 and CachyOS's 6.6.0-cbrt2.0 and -6.6.0-cb2.0 (both for Fedora via COPR), but the no success with the brightness controls (via Gnome or via my laptop's keys.
No /etc/nvidia.conf
On thing that I forgot that 545.29 also has some problem when building during boot. On Beta and now non-beta my machine won't finish booting if the module is not build before restarting with that kernel version with akmods. It requires shutdown and restart, then it will boot.

scaronni, 2023-11-15:
I've go some feedback that it works fine already, so i will push it to the main repository with some other changes. I've just moved the configuration file to `/etc/nvidia/kernel.conf` to be more compliant with other Nvidia products.

msmafra, 2023-11-15:
Hello
I did a fresh install of Fedora 39 for testing on a different SSD. The brightness still not working (mind you I use discrete GPU mode by default). The hang on boot after the installation is only happening on my current install, on the fresh one everything went smooth.

scaronni, 2023-11-16:
https://negativo17.org/nvidia-proprietary-and-open-source-kernel-modules/

msmafra, 2023-11-16:
Awesome! Thanks for the guide. Unfortunately it did not work on my system.

scaronni, 2023-11-23:
Fixed in 545.29.06, waiting for the open source bits to be released: https://www.nvidia.com/Download/driverResults.aspx/216530/en-us/

Packages will be online as soon as they are available.

closed 2023-11-23 by scaronni