negativo17/nvidia-driver #147: 545.29.06 Linux Driver fix update

DimPaDev, 2023-11-22:
Just wanted to let you know that NVIDIA  545.29.06 Linux Driver was released today! it may fix some issues and regressions. 

Thank you very much for providing this great repo and keep up the great work!

scaronni, 2023-11-23:
Ongoing! Thanks.

scaronni, 2023-11-23:
Talked too soon, the open source components have not been yet released:

- https://github.com/NVIDIA/nvidia-modprobe/tags
- https://github.com/NVIDIA/nvidia-persistenced/tags
- https://github.com/NVIDIA/nvidia-settings/tags
- https://github.com/NVIDIA/nvidia-xconfig/tags

scaronni, 2023-11-23:
Usually those are available once the announcement is done at this page: https://forums.developer.nvidia.com/c/gpu-graphics/announcements-and-news/146

scaronni, 2023-11-26:
> Talked too soon, the open source components have not been yet released:
> 
> - https://github.com/NVIDIA/nvidia-modprobe/tags
> - https://github.com/NVIDIA/nvidia-persistenced/tags
> - https://github.com/NVIDIA/nvidia-settings/tags
> - https://github.com/NVIDIA/nvidia-xconfig/tags

Nope, these above.

kmare, 2023-11-26:
Thanks @scaronni !

msmafra, 2023-12-01:
Fixed for me with 545.29.06. Thanks @scaronni .

closed 2023-12-20 by scaronni