negativo17/nvidia-driver #149: Unable to build kernel module for the 6.7.3 kernel.

vpxyz, 2024-02-06:
Hi, 
due to a bug in the common/inc/nv-linux.h file, it is not possible to compile the nvidia kernel module. Fortunately, there is a patch:
https://github.com/gentoo/gentoo/blob/c64caf53/x11-drivers/nvidia-drivers/files/nvidia-drivers-470.223.02-gpl-pfn_valid.patch
RPM Fusion has updated the akmod-nvidia package. 
More information here: 
https://forums.developer.nvidia.com/t/linux-6-7-3-545-29-06-550-40-07-error-modpost-gpl-incompatible-module-nvidia-ko-uses-gpl-only-symbol-rcu-read-lock/280908/29

Thanks

scaronni, 2024-02-06:
Hi @vpxyz by the time you wrote the ticket the RPMs were already pushed (545.29.06-2) in the repositories (DKMS/akmod for both Fedora 38 and 39), but thanks for reporting!

scaronni, 2024-02-06:
Let me know if it works, thanks.

closed 2024-02-06 by scaronni

vpxyz, 2024-02-06:
It works!
Thanks