negativo17/nvidia-driver #151: Accidentaly discoverd that egl-wayland is not being installed, so GDM won't work on Wayland sessions

msmafra, 2024-02-11:
Hello there!

I was doing a fresh installation of Fedora Silverblue and after Fedora Workstation for comparison and testing if I can migrate to Silverblue. For that I just installed few things including the nvidia drivers and did not run my script that installs everything I use or need, but after installing I couldn't log in anymore. GDM would freeze... after installing some packages, reinstalling nvidia drivers, thinking that was related to the current kernel or CachyOS's kernel, I removed Silverblue installed Workstation to have the same thing happen. Long story short, after a while I installed some mesa packages and egl-wayland, and it was the lack of egl-wayland that was causing the problem. 
To make sure, I tested with Silverblue and Workstation, since I was not sure if it was just egl-wayland. After both installations stop freezing as soon as I installed egl-wayland I confirmed it and also tested with RPMFusion. With RPMFusin egl-wayland is installed together with the nvidia drivers.
I saw on netgativo17.org that one of the packages is egl-wayland, so it's probably missing from the dependencies.

scaronni, 2024-02-16:
There is this in the history: https://github.com/negativo17/nvidia-driver/commit/cd6f2b9044d90f71f94fa91be1cc0cad343a1560

I don't remember exactly the context for this, but if I try to remove it now, there is no dependency on it. I will re add it as an explicit dependency.

Thanks for reporting!

scaronni, 2024-02-16:
Closed in https://github.com/negativo17/nvidia-driver/commit/85d426ccd8feef2836e3bf07bc163c13acc1a496

closed 2024-02-16 by scaronni

msmafra, 2024-02-16:
Thank you kindly