negativo17/nvidia-driver #155: epel8 driver v 550.76-1.el8 pulls in dkms-nvidia even though kmod-nvidia is used.

ElCoyote27, 2024-04-27:
Hi Simone,

I noticed there was a recent update for EPEL8 build of the nvidia driver.
As usual, since I am using RHEL8 EUS, I went to rebuild the nvidia kmod src.rpm (RHEL8 EUS is currently on RHEL 8.8 so your builds against 8.9 do not load and I have to rebuild).
On the previous version, dkms-nvidia was not a dependency:
```
# rpm -q kmod-nvidia
kmod-nvidia-550.67-1.8.el8.x86_64

```
But on 550.76-1, dkms-nvidia got pulled when I tried to update the kmod-nvidia rpm I had just rebuilt:

```
Dependencies resolved.
=============================================================================================================================================
 Package                               Architecture       Version                        Repository                                     Size
=============================================================================================================================================
Upgrading:
 kmod-nvidia                           x86_64             3:550.78-1.8.el8               @commandline                                   40 M
 nvidia-driver                         x86_64             3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms             2.5 M
 nvidia-driver-NVML                    i686               3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms             575 k
 nvidia-driver-NVML                    x86_64             3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms             596 k
 nvidia-driver-NvFBCOpenGL             i686               3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms              60 k
 nvidia-driver-NvFBCOpenGL             x86_64             3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms              54 k
 nvidia-driver-cuda                    x86_64             3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms             2.2 M
 nvidia-driver-cuda-libs               i686               3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms              44 M
 nvidia-driver-cuda-libs               x86_64             3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms              44 M
 nvidia-driver-libs                    i686               3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms              33 M
 nvidia-driver-libs                    x86_64             3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms             146 M
 nvidia-kmod-common                    noarch             3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms              36 M
 nvidia-libXNVCtrl                     x86_64             3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms              28 k
 nvidia-libXNVCtrl-devel               x86_64             3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms              58 k
 nvidia-persistenced                   x86_64             3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms              40 k
 nvidia-settings                       x86_64             3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms             806 k
Installing dependencies:
 dkms-nvidia                           x86_64             3:550.76-1.el8                 krynn_infra_Negativo17_rhel8-rpms              45 M

```
That is a little surprising since your .spec bears no mention of dkms:

I built this while I had dkms 3.0.12 installed, perhaps that is why. I will try a rebuild/

Thank you for all the work on these NVidia drivers. It r

ElCoyote27, 2024-04-27:
There was most likely an implicit dependency:

```
# rpm -e dkms-nvidia
error: Failed dependencies:
        nvidia-kmod = 3:550.76 is needed by (installed) nvidia-kmod-common-3:550.76-1.el8.noarch

```

ElCoyote27, 2024-04-27:
550.78 is out, double checking and will re-open if needed.

closed 2024-04-27 by ElCoyote27

scaronni, 2024-04-27:
Hi, both `dkms-nvidia` and `kmod-nvidia` provide `nvidia-kmod`, so whatever you install the dependency of `nvidia-kmod-common` is satisfied:

```
$ rpm -qp --provides kmod-nvidia-550.78-1.el8.x86_64.rpm | grep ^nvidia-kmod
nvidia-kmod = 3:550.78-1.el8
$ rpm -qp --provides dkms-nvidia-550.78-1.el8.x86_64.rpm | grep ^nvidia-kmod
nvidia-kmod = 3:550.78
$ rpm -qp --requires nvidia-kmod-common-550.78-1.el8.noarch.rpm | grep ^nvidia-kmod
nvidia-kmod = 3:550.78
```

Why the dependency was not satisfied by your rebuild of `kmod-nvidia` is a good question, it's hardcoded in the SPEC file. Just out of curiosity, why don't you use `dkms-nvidia` if in the end you always need to rebuild the kernel module package?

scaronni, 2024-04-27:
Ah, I see now, I missed on first look. It's because you're installing a kernel module for a different version.

You're building `kmod-nvidia` at version 550.78 while the rest of the packages are at 550.76. Without specifying anything, the first package that solves `nvidia-kmod = 3:550.76` is selected, which is `dkms-nvidia`.

ElCoyote27, 2024-04-27:
Hi Simone, yes, I realized my mistake shortly after that.
My Satellite had mirrored and published my el8_hvms CV with 550.76 but I had in fact rebuilt the kmod for 550.78. Once I realized that mistake everything was OK. Sorry for the noise.
I must rebuild kmod-nvidia when EUS is on a different minor version (e.g: 8.8) as the 'current' (8.9) version because the 8.9 kABI is often not downward compatible.
Thank you!