negativo17/nvidia-driver #156: nvidia-driver.spec is missing a 32-bit library essential for embedded gamescope/steam

matte-schwartz, 2024-05-25:
Contuing from here: https://github.com/ValveSoftware/steam-for-linux/issues/10916#issuecomment-2129975306

Basically, it seems like the `nvidia-driver.spec` doesn't include at least one essential 32-bit library for embedded gamescope and steam to work properly on 555.42.02, `nvidia-drm_gbm.so`. 

according to the Nvidia dev's comments in my issue, `/usr/lib/gbm/nvidia-drm_gbm.so` should be symlinked to `/usr/lib/libnvidia-allocator.so.1`, while it appears that the .spec only does this for x86_64 and aarch64 build architectures. I had to manually create the `gbm` folder in my `/usr/lib` directory and then symlink the file.

there may be other libraries missing, but this one tweak was enough to get embedded gamescope and steam not to crash.

scaronni, 2024-05-27:
Thanks for reporting, will add immediately! I guess that by trying to use `gamescope` you are trying to mimick a steam deck like experience.

Do you have any information to share for `gamescope` etc. on Fedora? I was providing the old `steamos` packages back in the day to have a console like experience with Fedora and Big Picture, but for a few years I did not have enough hardware to play with, so I dropped it when the original SteamOS component got dropped.

Now that I work for Nvidia I have a few more options and I would like to get back to it.

scaronni, 2024-05-27:
By chance, do you also have `egl-gbm.i686` installed beside needing the symlink? So far that 32 bit package has not been a dependency.

closed 2024-05-27 by scaronni

matte-schwartz, 2024-05-28:
Thanks for taking care of this so swiftly! For what its worth, even the official NVIDIA .run file is incorrect on this... I've let the dev know who told me about the symlink in the first place after I confirmed in a VM.

As far as gamescope on Fedora goes, I help maintain Nobara and we build gamescope from Valve's master branch very regularly along with shipping slightly patched versions of  [gamescope-session-plus](https://github.com/ChimeraOS/gamescope-session) and [gamescope-session-steam](https://github.com/ChimeraOS/gamescope-session-steam) from ChimeraOS. We provide some gamescope patches that take care of some minor things like changing gamescope's default resolution in gamescope-session from 720p (deck resolution) to the native display resolution. 

Along with `jupiter-hw-support` from Valve as a dependency for gamescope-session, this combination of packages gives you feature parity to SteamOS aside from a few minor Deck-specific details, like the TDP slider available in the Deck client's Steam quick menu. All of this is available on our GitHub if you'd like to take a look at how its packaged: https://github.com/Nobara-Project/rpm-sources

In terms of embedded gamescope and steam compatibility with Nvidia, there are two issues still need to be addressed now the Nvidia drivers don't cause the two to coredump immediately, and both are a bit out of my grasp unfortunately. 

The first issue is perhaps a driver or gamescope issue where HDR colorspace seems to cause massive display corruption throughout the Steam UI and games, even if they're not currently running in HDR. I've posted a video of it in that thread since I seem to be the only person to have access to HDR setups to test on in there: https://github.com/ValveSoftware/gamescope/issues/1320#issuecomment-2131405485

The second issue is most likely a Steam issue where the Steam UI both inside of gamescope and outside of gamescope (Big Picture Mode) have garbled side menus when you pull them up: 
[image]

I plan to keep being active in that gamescope thread, along with continuing my normal testing of gamescope versions as they come out for any changes or regressions. Happy to provide more input or answer any questions if you have any about all of this.

scaronni, 2024-05-28:
Thanks for all the info!

> Along with `jupiter-hw-support` from Valve as a dependency for gamescope-session, this combination of packages gives you feature parity to SteamOS aside from a few minor Deck-specific details, like the TDP slider available in the Deck client's Steam quick menu.

Why the dependency on `jupiter-hw-support` even if not running on the Steam Deck hardware?

> All of this is available on our GitHub if you'd like to take a look at how its packaged: https://github.com/Nobara-Project/rpm-sources

I've seen, there's also a checkout as-is of my packaged drivers, maybe you can just use my repositories instead (up to you, doesn't really matter for me).
 
> The second issue is most likely a Steam issue where the Steam UI both inside of gamescope and outside of gamescope (Big Picture Mode) have garbled side menus when you pull them up:

I don't have garbled menus in Steam, at least context menus, main manu, etc. Are you using 555 drivers?

Vash63, 2024-08-21:
> Thanks for all the info!
>
>
> Why the dependency on `jupiter-hw-support` even if not running on the Steam Deck hardware?
>
>
> I've seen, there's also a checkout as-is of my packaged drivers, maybe you can [...]

Make sure you enable GPU accelerated rendering in web views under Steam Settings -> Interface, this exposes some of the corruption issues Matte  reported. They've been improved over recent NV driver releases but I still see them sometimes when mousing over some menus or right clicking in friends sometimes.