negativo17/nvidia-driver #157: Help understanding versioning and branching

p5, 2024-08-09:
Hey there!

Thank you for packaging these drivers, we have recently switched to these builds from RPMFusion for our project, and they are working great!

It would be good to get some clarification on the versioning though.  The documentation on your website suggests the current supported Fedora versions (as of now, 39 and 40) should use the "short lived branch version" while rawhide contains the "beta branch version".

> The CentOS/RHEL repository contains the “Long Lived Branch version” where less changes occur; while Fedora repositories contains the “Short Lived Branch version”. Beta CentOS/RHEL and Fedora’s rawhide repositories will contain the “Beta Branch version”

I may be misunderstanding something, but I interpreted that as Fedora 39 & 40 should still be on version 555, since 560 is currently in beta according to [Nvidia's website](https://www.nvidia.com/en-us/drivers/unix/).  

It would be great to have some clarification on when these updates should be provided, as being on the absolute latest available drivers for your daily driver can cause a few unexpected issues.

Look forward to your response

Thanks
Rob

scaronni, 2024-08-09:
> It would be good to get some clarification on the versioning though. The documentation on your website suggests the current supported Fedora versions (as of now, 39 and 40) should use the "short [...]

Actually since I last updated that page, Nvidia changed the wording. Omitting the LTS branches that are still updated (535 for example) it's now with the [following](https://www.nvidia.com/en-us/drivers/unix/):

- Latest Production Branch (ex Long Lived)
- Latest New Feature Branch (ex Short Lived)
- Latest Beta

I usually try to follow that, mapping them with EL / Fedora / Rawhide, but sometimes I do exceptions.

In this particular case there was the whole hype and expectations for proper performance on Wayland with Explicit Sync, and it began with 555 and ended in 560 with full support for Vulkan and the updated EGL libraries.

We've just released 560.28.03 as a fully supported version in the [CUDA repository](https://developer.download.nvidia.com/compute/cuda/repos/) also for RHEL, regardless of the beta status. As of [today](https://github.com/NVIDIA/egl-wayland/pull/125) also the EGL libraries are finished so they will be pushed to [Fedora / EPEL](https://bodhi.fedoraproject.org/updates/?search=egl) as well as in the CUDA repository.

I've updated driver 560 for RHEL/Fedora in the CUDA repository to be almost in sync with my repository here, with the exception of the boot update script and the AppStream metadata, which is hard to get into the CUDA repository due to the DNF module usage.

closed 2024-08-09 by scaronni

p5, 2024-08-10:
Awesome!  Thank you for the clarification.

I have just found out you're keeping the older drivers around, so it's trivial to manage the updates on our own timeline.

For me and the project I am involved with, remaining pinned to the latest feature version is key, which the Negativo drivers allow.  So I'm happy 😄

scaronni, 2024-10-26:
@p5 another example fresh for today: 565 is considered beta (not internally actually) but the driver contains basically only bugfixes (lots of them!), Blackwell support and is actually the next one after 560 that fixes this vulnerability: https://nvidia.custhelp.com/app/answers/detail/a_id/5586/~/security-bulletin%3A-nvidia-gpu-display-driver---october-2024

No information if 560 will ever be fixed. So I'm pushing this also to EPEL and Fedora branches.

scaronni, 2024-10-26:
Actually no, 560 will not be fixed. so 565 is the only option.