negativo17/nvidia-driver #158: 560.31 does not work with GPUs not supported by open nvidia.ko

adatum, 2024-08-14:
After upgrading to 560.31.x, the nvidia driver does not load, and system logs contain the following message:

```
kernel: NVRM: The NVIDIA GPU 0000:09:00.0 (PCI ID: 10de:1d01)
             NVRM: installed in this system is not supported by open
             NVRM: nvidia.ko because it does not include the required GPU
             NVRM: System Processor (GSP).
             NVRM: Please see the 'Open Linux Kernel Modules' and 'GSP
             NVRM: Firmware' sections in the driver README, available on
             NVRM: the Linux graphics driver download page at
             NVRM: www.nvidia.com.
kernel: nvidia 0000:09:00.0: probe with driver nvidia failed with error -1
kernel: NVRM: The NVIDIA probe routine failed for 1 device(s).
kernel: NVRM: None of the NVIDIA devices were initialized.
```

GPU is a GT 1030.

Downgrading to 555.58.02 and pinning this version is a workaround for now. What is the proper way forward?

mwawra, 2024-08-14:
Hi @adatum , your GPU is not compatible with the new open source kernel module of the NVIDIA driver (as mine). You can switch back to proprietary version:
`# sed -i -e 's/kernel-open$/kernel/g' /etc/nvidia/kernel.conf
# akmods --rebuild
# reboot`

https://negativo17.org/nvidia-driver/#Proprietary_and_open_source_kernel_modules
https://developer.nvidia.com/blog/nvidia-transitions-fully-towards-open-source-gpu-kernel-modules/

scaronni, 2024-08-14:
thanks @mwawra. Also here I made a post in November: https://negativo17.org/nvidia-proprietary-and-open-source-kernel-modules/

mwawra, 2024-08-14:
Hi @scaronni, even though I am not sure if there is a good solution for this, I think this is a bug and the post should perhaps not be closed yet. Especially since even with existing installations the kernel driver is changed to the open source driver and thus a normal update makes the current configuration unusable for older GPUs. This also means, if I understand it correctly, that I have to change it again after every further update. Or was this only the case with this update?

scaronni, 2024-08-14:
If you check the rpm, the file is marked as a configuration file, so from now on, if you have customized the file, it will stay like this and the new configuration file will be installed as `.rpmnew`.

scaronni, 2024-08-14:
Actually duplicate of https://github.com/negativo17/nvidia-kmod-common/issues/13

adatum, 2024-08-14:
Thanks @mwawra and @scaronni . Changing back to the proprietary `kernel` module, instead of `kernel-open`, indicated in `/etc/nvidia/kernel.conf` file worked perfectly.

Now I wonder what happens to support for older GPUs when the open source kernel module becomes the only option? I really wish I could use the open source kernel module with my card.

scaronni, 2024-08-14:
At one point old cards will be supported only by legacy branches. When no card requiring proprietary modules will be in main branches, the proprietary modules will be removed.

adatum, 2024-08-14:
@scaronni Do you also package legacy drivers in the negativo repo? Hopefully the need for legacy drivers is a long way off. By then I could look into alternatives, but it sure has been great upgrading drivers mindlessly with the negativo repo. Many thanks for your long-standing efforts!

scaronni, 2024-08-16:
No plan at the moment. Stuff that is supported by the legacy releases only are 10+ years old cards, and in most cases the driver it's just not compatible with changes in the distributions.

I might package one of the "datacenter" drivers that are still being updated even if not explicitly listed as legacy, beta, production etc. Branch 535 is currently the one that falls in this category: https://www.nvidia.com/en-us/drivers/unix/linux-amd64-display-archive/

As you can see, even if released in May 2023 is still being updated.

brittyazel, 2024-10-10:
> If you check the rpm, the file is marked as a configuration file, so from now on, if you have customized the file, it will stay like this and the new configuration file will be installed as `.rpmnew`.

I don't believe this file is actually marked as such, my kernel.conf files are getting overwritten on package updates with my changes getting moved to a .rpmsave file. I opened up a new issue for this [here](
https://github.com/negativo17/nvidia-kmod-common/issues/14).

`[Afaik](https://access.redhat.com/solutions/60263)`, this file needs to be marked as `%config(noreplace)` in the .spec, but currently it is only marked as `%config`

scaronni, 2024-10-11:
@brittyazel thanks for reporting, I've fixed it and the new packages will be online shortly.