negativo17/nvidia-driver #160: DNF and DNF5 failing to install xorg-x11-nvidia-560.35.03-2.fc40.x86_64.rpm "The package is not signed."/GPG check failed

msmafra, 2024-08-31:
When running a dnf upgrade/distro-sync nvidia drivers are not updating. 
It seems that only the file "xorg-x11-nvidia-560.35.03-2.fc40.x86_64.rpm" is the problem.

[image]

[image]

my4ng, 2024-08-31:
Seems like more (if not all) packages are affected:

```
Package xorg-x11-nvidia-560.35.03-2.fc40.x86_64.rpm is not signed
Package libnvidia-cfg-560.35.03-2.fc40.x86_64.rpm is not signed
Package libnvidia-ml-560.35.03-2.fc40.x86_64.rpm is not signed
Package nvidia-driver-560.35.03-2.fc40.x86_64.rpm is not signed
Package nvidia-driver-cuda-560.35.03-2.fc40.x86_64.rpm is not signed
Package nvidia-driver-cuda-libs-560.35.03-2.fc40.x86_64.rpm is not signed
Package nvidia-driver-libs-560.35.03-2.fc40.x86_64.rpm is not signed
The downloaded packages were saved in cache until the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: GPG check FAILED
```

msmafra, 2024-08-31:
Yep!

```
Downloading Packages:
[SKIPPED] dkms-3.0.13-1.fc40.noarch.rpm: Already downloaded                                              
[SKIPPED] nvidia-gpu-firmware-20240811-2.fc40.noarch.rpm: Already downloaded                             
[SKIPPED] libnvidia-ml-560.35.03-2.fc40.i686.rpm: Already downloaded                                     
[SKIPPED] libnvidia-ml-560.35.03-2.fc40.x86_64.rpm: Already downloaded                                   
[SKIPPED] nvidia-driver-cuda-560.35.03-2.fc40.x86_64.rpm: Already downloaded                             
(6/15): nvidia-driver-560.35.03-2.fc40.x86_64.rpm                         992 kB/s | 3.9 MB     00:03    
(7/15): nvidia-driver-cuda-libs-560.35.03-2.fc40.i686.rpm                 3.8 MB/s |  40 MB     00:10    
(8/15): dkms-nvidia-560.35.03-1.fc40.x86_64.rpm                           3.7 MB/s |  55 MB     00:14    
(9/15): nvidia-driver-libs-560.35.03-2.fc40.i686.rpm                      6.6 MB/s |  31 MB     00:04    
(10/15): nvidia-libXNVCtrl-560.35.03-1.fc40.x86_64.rpm                    110 kB/s |  25 kB     00:00    
(11/15): nvidia-modprobe-560.35.03-1.fc40.x86_64.rpm                      136 kB/s |  31 kB     00:00    
(12/15): nvidia-driver-cuda-libs-560.35.03-2.fc40.x86_64.rpm              3.3 MB/s |  40 MB     00:12    
(13/15): nvidia-settings-560.35.03-1.fc40.x86_64.rpm                      1.8 MB/s | 796 kB     00:00    
(14/15): nvidia-xconfig-560.35.03-1.fc40.x86_64.rpm                       414 kB/s |  92 kB     00:00    
(15/15): nvidia-driver-libs-560.35.03-2.fc40.x86_64.rpm                   5.4 MB/s | 150 MB     00:27    
----------------------------------------------------------------------------------------------------------
Total                                                                     7.6 MB/s | 320 MB     00:42     
Package libnvidia-ml-560.35.03-2.fc40.i686.rpm is not signed
Package libnvidia-ml-560.35.03-2.fc40.x86_64.rpm is not signed
Package nvidia-driver-560.35.03-2.fc40.x86_64.rpm is not signed
Package nvidia-driver-cuda-560.35.03-2.fc40.x86_64.rpm is not signed
Package nvidia-driver-cuda-libs-560.35.03-2.fc40.i686.rpm is not signed
Package nvidia-driver-cuda-libs-560.35.03-2.fc40.x86_64.rpm is not signed
Package nvidia-driver-libs-560.35.03-2.fc40.i686.rpm is not signed
Package nvidia-driver-libs-560.35.03-2.fc40.x86_64.rpm is not signed
The downloaded packages were saved in cache until the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: GPG check FAILED
```

scaronni, 2024-09-01:
Sorry, put a reminder to replace the GPG used to sign packages with a stronger one after the expiration of EL7 and in the end I forgot to do that. Will be fixed in a few minutes: https://github.com/negativo17/gpg-keys

scaronni, 2024-09-01:
Repositories are currently being synced.

scaronni, 2024-09-01:
https://negativo17.org/gpg-key-rotation/

closed 2024-09-01 by scaronni