negativo17/nvidia-driver #161: Fedora 40 KDE Spin Nvidia 560 driver

geosone, 2024-09-11:
With the newest 560 driver the system starts shows shortly the curser then goes back to black.

do you have any idee.

geosone, 2024-09-11:
ok found it has nothing to dod with the driver but with other update

`Sep 11 07:44:07 amorine systemd[1]: Started session-c1.scope - Session c1 of User sddm.
Sep 11 07:44:07 amorine audit[2698]: USER_START pid=2698 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:xdm_t:s0-s0:c0.c1023 msg='op=PAM:session_open grantors=pam_unix,pam_systemd acct="sddm" exe=>
Sep 11 07:44:10 amorine sddm-greeter-qt6[2733]: Loading file:///usr/share/sddm/themes/01-breeze-fedora/Main.qml...
Sep 11 07:44:11 amorine audit[2733]: ANOM_ABEND auid=4294967295 uid=988 gid=986 ses=4294967295 subj=system_u:system_r:xdm_t:s0-s0:c0.c1023 pid=2733 comm="QQuickPixmapRea" exe="/usr/bin/sddm-greeter-qt6" sig=4 >
Sep 11 07:44:11 amorine systemd-coredump[2840]: [🡕] Process 2733 (sddm-greeter-qt) of user 988 dumped core.
                                                
                                                Module libopenjph.so.0.15 from rpm openjph-0.15.0-6.fc40.x86_64
                                                Module libheif-jphenc.so from rpm libheif-1.18.2-2.fc40.x86_64
                                                Module libheif-jpegenc.so from rpm libheif-1.18.2-2.fc40.x86_64
                                                Module libheif-jpegdec.so from rpm libheif-1.18.2-2.fc40.x86_64
                                                Module libheif-j2kenc.so from rpm libheif-1.18.2-2.fc40.x86_64
                                                Module libheif-j2kdec.so from rpm libheif-1.18.2-2.fc40.x86_64
                                                Module libdatrie.so.1 from rpm libdatrie-0.2.13-9.fc40.x86_64
                                                Module libthai.so.0 from rpm libthai-0.1.29-8.fc40.x86_64
                                                Module libfribidi.so.0 from rpm fribidi-1.0.14-2.fc40.x86_64
                                                Module libgpg-error.so.0 from rpm libgpg-error-1.49-1.fc40.x86_64
                                                Module libnuma.so.1 from rpm numactl-2.0.16-5.fc40.x86_64
                                                Module liblpcnetfreedv.so.0.5 from rpm lpcnetfreedv-0.5-5.fc40.x86_64
                                                Module libpixman-1.so.0 from rpm pixman-0.43.4-1.fc40.x86_64
                                                Module libxcb-shm.so.0 from rpm libxcb-1.17.0-1.fc40.x86_64
                                                Module libXrender.so.1 from rpm libXrender-0.9.11-6.fc40.x86_64
                                                Module libpango-1.0.so.0 from rpm pango-1.54.0-1.fc40.x86_64
                                                Module libpangoft2-1.0.so.0 from rpm pango-1.54.0-1.fc40.x86_64
                                                Module libpangocairo-1.0.so.0 from rpm pango-1.54.0-1.fc40.x86_64
                                                Module libgdk_pixbuf-2.0.so.0 from rpm gdk-pixbuf2-2.42.10-8.fc40.x86_64
                                                Module libcairo-gobject.so.2 from rpm cairo-1.18.0-3.fc40.x86_64
                                                Module libOpenCL.so.1 from rpm ocl-icd-2.3.2-6.fc40.x86_64
                                                Module libvdpau.so.1 from rpm libvdpau-1.5-11.fc40.x86_64
                                                Module libva-x11.so.2 from rpm libva-2.22.0-1.fc40.x86_64
                                                Module libva-drm.so.2 from rpm libva-2.22.0-1.fc40.x86_64
                                                Module libvpl.so.2 from rpm libvpl-2.13.0-1.fc40.x86_64
                                                Module libva.so.2 from rpm libva-2.22.0-1.fc40.x86_64
                                                Module libkvazaar.so.7 from rpm kvazaar-2.3.1-1.fc40.x86_64
                                                Module libopenh264.so.7 from rpm openh264-2.4.1-1.fc40.x86_64
                                                Module libxvidcore.so.4 from rpm xvidcore-1.3.7-11.fc40.x86_64
                                                Module libxavs2.so.13 from rpm xavs2-1.3-5.20190422giteae1e8b.fc40.x86_64
                                                Module libxavs.so.1 from rpm xavs-0.1.55-1.fc40.x86_64
                                                Module libSvtHevcEnc.so.1 from rpm svt-hevc-1.5.1-6.fc40.x86_64
                                                Module libx265.so.209 from rpm x265-3.6-10.fc40.x86_64
                                                Module libx264.so.164 from rpm x264-0.164-34.20240513git4613ac3.fc40.x86_64
                                                Module libSvtVp9Enc.so.1 from rpm svt-vp9-0.3.0-11.fc40.x86_64
                                                Module libvorbisenc.so.2 from rpm libvorbis-1.3.7-10.fc40.x86_64
                                                Module libtwolame.so.0 from rpm twolame-0.4.0-4.fc40.x86_64
                                                Module libtheoradec.so.1 from rpm libtheora-1.1.1-36.fc40.x86_64
                                                Module libtheoraenc.so.1 from rpm libtheora-1.1.1-36.fc40.x86_64
                                                Module libspeex.so.1 from rpm speex-1.2.0-17.fc40.x86_64
                                                Module libopus.so.0 from rpm opus-1.5.1-1.fc40.x86_64
                                                Module libopenjp2.so.7 from rpm openjpeg2-2.5.2-1.fc40.x86_64
[... cut ...]
                                                #19 0x00007f0d984ad973 _ZN10QQuickView15continueExecuteEv (libQt6Quick.so.6 + 0x2ad973)
                                                #20 0x000055f8850120f0 _ZN4SDDM10GreeterApp16addViewForScreenEP7QScreen (sddm-greeter-qt6 + 0x270f0)
                                                #21 0x000055f8850129ac _ZN4SDDM10GreeterApp11customEventEP6QEvent (sddm-greeter-qt6 + 0x279ac)
                                                #22 0x00007f0d967edd5f _ZN7QObject5eventEP6QEvent (libQt6Core.so.6 + 0x1edd5f)
`

after removing libopenjph

and its dependeices sddm is working again.

closed 2024-09-11 by geosone

scaronni, 2024-09-12:
Yes, there was a bug on it in Fedora's Bugzilla: https://bugzilla.redhat.com/show_bug.cgi?id=2307795

TL;DR: every time you change the instructions set you enable in OpenJPH, even if the version does not change, require a rebuild of `libheif`. For a moment there was a newer OpenJPH in `updates-testing` than my multimedia repository as I intend to phase out my copy.

I've rebuild `libheif`, so the bug is fixed.