negativo17/nvidia-driver #162: Transaction test error with nvidia-driver-3:560.35.03-1.el9.x86_64

cbsrobot, 2024-09-12:
Installing nvidia-driver-3:560.35.03-1.el9.x86_64 on Rocky 9 gives the following error:

```
$ sudo dnf module install nvidia-driver:open-dkms 
[...]
Error: Transaction test error:
  file /etc/X11/xorg.conf.d/10-nvidia.conf conflicts between attempted installs of xorg-x11-nvidia-3:560.35.03-2.el9.x86_64 and nvidia-driver-3:560.35.03-1.el9.x86_64
```

diffing the two files it seems there is a minor change:
```
diff /tmp/xorg-x11-nvidia-560.35.03-2.el9.x86_64.rpm/etc/X11/xorg.conf.d/10-nvidia.conf /tmp/nvidia-driver-560.35.03-1.el9.x86_64.rpm/etc/X11/xorg.conf.d/10-nvidia.conf
5a6
>     Option "PrimaryGPU" "yes"
```

cbsrobot, 2024-09-12:
```
$ cat /tmp/nvidia-driver-560.35.03-1.el9.x86_64.rpm/etc/X11/xorg.conf.d/10-nvidia.conf 
Section "OutputClass"
    Identifier "nvidia"
    MatchDriver "nvidia-drm"
    Driver "nvidia"
    Option "AllowEmptyInitialConfiguration"
    Option "PrimaryGPU" "yes"
    Option "SLI" "Auto"
    Option "BaseMosaic" "on"
EndSection
```

scaronni, 2024-09-15:
Hello, you are using the CUDA repository along with my driver, the conflicting changes are between the package in my repository and the official CUDA one. I've forgot to have a newer release in my repository to override the official package.

The same changes will be in the ~570 Nvidia driver releases.

I'll do in a minute. Thanks for reporting.

closed 2024-09-15 by scaronni