negativo17/nvidia-driver #164: Getting different NVIDIA versions/builds from fedora-nvidia and fedora-multimedia

msmafra, 2024-10-10:
Hello there

Did not notice it before, just now on Fedora 41 Beta trying to solve multimedia related conflicts with fedora-updates, but it probably happen few weeks back even when I was on Fedora 40: The fedora-nvidia repo has a different "version/build" of the NVIDIA drivers than fedora-multimedia.

```
Package                                                   Arch          Version                                                   Repository                            Size
Upgrading:
 libnvidia-cfg                                            x86_64        3:560.35.03-4.fc41                                        fedora-multimedia                393.7 KiB
   replacing libnvidia-cfg                                x86_64        3:560.35.03-3.fc41                                        @commandline                     393.7 KiB
 libnvidia-fbc                                            i686          3:560.35.03-4.fc41                                        fedora-multimedia                322.3 KiB
   replacing libnvidia-fbc                                i686          3:560.35.03-3.fc41                                        fedora-nvidia                    322.3 KiB
 libnvidia-fbc                                            x86_64        3:560.35.03-4.fc41                                        fedora-multimedia                311.9 KiB
   replacing libnvidia-fbc                                x86_64        3:560.35.03-3.fc41                                        fedora-nvidia                    311.9 KiB
 libnvidia-ml                                             i686          3:560.35.03-4.fc41                                        fedora-multimedia                  2.1 MiB
   replacing libnvidia-ml                                 i686          3:560.35.03-3.fc41                                        fedora-nvidia                      2.1 MiB
 libnvidia-ml                                             x86_64        3:560.35.03-4.fc41                                        fedora-multimedia                  2.0 MiB
   replacing libnvidia-ml                                 x86_64        3:560.35.03-3.fc41                                        fedora-nvidia                      2.0 MiB
 nvidia-driver                                            x86_64        3:560.35.03-4.fc41                                        fedora-multimedia                 15.9 MiB
   replacing nvidia-driver                                x86_64        3:560.35.03-3.fc41                                        fedora-nvidia                     15.9 MiB
 nvidia-driver-cuda                                       x86_64        3:560.35.03-4.fc41                                        fedora-multimedia                  1.3 MiB
   replacing nvidia-driver-cuda                           x86_64        3:560.35.03-3.fc41                                        fedora-nvidia                      1.3 MiB
 nvidia-driver-cuda-libs                                  i686          3:560.35.03-4.fc41                                        fedora-multimedia                175.2 MiB
   replacing nvidia-driver-cuda-libs                      i686          3:560.35.03-3.fc41                                        fedora-nvidia                    175.2 MiB
 nvidia-driver-cuda-libs                                  x86_64        3:560.35.03-4.fc41                                        fedora-multimedia                192.1 MiB
   replacing nvidia-driver-cuda-libs                      x86_64        3:560.35.03-3.fc41                                        fedora-nvidia                    192.1 MiB
 nvidia-driver-libs                                       i686          3:560.35.03-4.fc41                                        fedora-multimedia                130.1 MiB
   replacing nvidia-driver-libs                           i686          3:560.35.03-3.fc41                                        fedora-nvidia                    130.1 MiB
 nvidia-driver-libs                                       x86_64        3:560.35.03-4.fc41                                        fedora-multimedia                383.3 MiB
   replacing nvidia-driver-libs                           x86_64        3:560.35.03-3.fc41                                        fedora-nvidia                    383.3 MiB
 xorg-x11-nvidia                                          x86_64        3:560.35.03-4.fc41                                        fedora-multimedia                 19.3 MiB
   replacing xorg-x11-nvidia                              x86_64        3:560.35.03-3.fc41                                        fedora-multimedia                 19.3 MiB
```

scaronni, 2024-10-11:
Hi @msmafra I build and push to both repositories usually at the same time. There might be moments where something is being built for the multimedia repository (for example a rebase that require multiple packages to be rebuilt before pushing an update) that is holding back the updates for a while but eventually they will be in sync.

Regarding Fedora 41, I haven't finished building all the packages for it since I've been terribly busy at work, I hope to finish this weekend.

msmafra, 2024-10-11:
> Hi @msmafra I build and push to both repositories usually at the same time. There might be moments where something is being built for the multimedia repository (for example a rebase that require [...]
Oh, Ok. At least on my system it seems to have not affect anything. First I thought was something related to DNF5, but DNF4 did the same thing.

> Regarding Fedora 41, I haven't finished building all the packages for it since I've been terribly busy at work, I hope to finish this weekend.

I was kinda jump to switch to F41 to test mostly Tuned behavior.

closed 2024-10-11 by scaronni