negativo17/nvidia-driver #165: DKMS module disappeared after minor driver update

brittyazel, 2024-10-16:
I'm not sure what happened exactly, but after updating to the 560 driver a few days ago, I had to edit the /etc/nvidia/kernel.conf file to change back to the closed-source kernel driver. This also required rebuilding and reinstalling the DKMS module, which I did with these commands:

sudo dkms build -m nvidia/560.35.03 --force
sudo dkms install -m nvidia/560.35.03 --force

This worked great, that is until I ran a dnf update today and pulled in a minor update to dkms-nvidia, which seemingly deleted my dkms module entirely. After installing this package and rebooting, the dkms module is complete gone, and running the above two commands is necessary to get the module to work again. Likewise, if I manually do a dnf reinstall of the dkms-nvidia package, it happens again, the dkms module ends up completely gone.

Have you seen this before? I really don't want to have to manually run these commands each time dkms-nvidia is updated. This never used to be an issue, so I don't know why this is happening now.

scaronni, 2024-10-16:
Can you try with the latest update that I just pushed now? It fixes this: https://github.com/negativo17/dkms-nvidia/issues/12

Also, the last update before the one currently published that fixes the bug above also correctly marked the configuration file `/etc/nvidia/kernel.conf` as `config(noreplace)`, so you should not have to edit that file anymore.

Sorry for that, but I don't have any more legacy GPUs and I couldn't spot it.

closed 2024-10-16 by scaronni

brittyazel, 2024-10-16:
Yep, I can confirm installing the newest (fixed) package twice results in the issue being resolved. I'll run the reinstall twice on each of our affected systems thanks!