negativo17/nvidia-driver #166: [Not a issue] Not sure if this patch change may influence anything related to negativo17's nvidia driver

msmafra, 2024-11-06:
While was kinda whining and rambling about previously having problems with Tuned on Fedora, Leigh Scott) let me know that some NVIDIA parameters previously used on GRUB_CMDLINE_LINUX, are now set internally https://discussion.fedoraproject.org/t/tuned-causing-strange-behaviors/132997/28 
https://github.com/rpmfusion/nvidia-kmod/blob/master/make_modeset_default.patch

scaronni, 2024-11-09:
That patch switches the default settings for the kernel module to avoid the need to pass `modeset=1 fbdev=1`.

I'm setting them in a modprobe file, so that's basically the same:

```
$ cat /etc/modprobe.d/nvidia-modeset.conf 
# Nvidia modesetting support. Set to 0 or comment to disable kernel modesetting
# support. This must be disabled in case of SLI Mosaic.

options nvidia-drm modeset=1 fbdev=1
```

closed 2024-11-09 by scaronni

scaronni, 2024-11-10:
Will come soon: https://forums.developer.nvidia.com/t/wayland-support-for-the-565-release-series/312688