negativo17/nvidia-driver #168: several GPUs that have worked for years on RHEL8 suddenly are no longer supported by the Negativo17 driver. of

ElCoyote27, 2024-12-08:
I have a few GPUs in my RHEL8 systems using the Negativo17 drivers.
As of 565 (I think this was 565) several of my GPUs fail to load the driver and report:

```
[root@palanthas ~]# modprobe nvidia
modprobe: ERROR: could not insert 'nvidia': Device or resource busy
[root@palanthas ~]# modinfo nvidia -F version
565.57.01
[root@palanthas ~]# dmesg |grep nvidia|tail -10
               NVRM: nvidia.ko because it does not include the required GPU
               NVRM: www.nvidia.com.
[79742.397145] nvidia: probe of 0000:04:00.0 failed with error -1
[79742.397506] nvidia-nvlink: Unregistered Nvlink Core, major device number 508
[79742.475359] nvidia-nvlink: Nvlink Core is being initialized, major device number 508
[79742.477393] nvidia 0000:04:00.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
               NVRM: nvidia.ko because it does not include the required GPU
               NVRM: www.nvidia.com.
[79742.477586] nvidia: probe of 0000:04:00.0 failed with error -1
[79742.478049] nvidia-nvlink: Unregistered Nvlink Core, major device number 508

```
Casualties (no longer working):
- Nvidia Quadro P2200
- Geforce 1030GT

Still working:
- Nvidia RTX A4000

```
[root@palanthas ~]# rpm -qa|grep nvidia
libnvidia-cfg-565.57.01-6.el8.x86_64
nvidia-driver-libs-565.57.01-6.el8.i686
nvidia-libXNVCtrl-devel-565.57.01-2.el8.x86_64
libnvidia-ml-565.57.01-6.el8.i686
libnvidia-fbc-565.57.01-6.el8.x86_64
nvidia-driver-cuda-565.57.01-6.el8.x86_64
pcp-pmda-nvidia-gpu-5.3.7-22.el8_10.x86_64
dkms-nvidia-565.57.01-2.el8.x86_64
nvidia-xconfig-565.57.01-2.el8.x86_64
nvidia-persistenced-565.57.01-2.el8.x86_64
libnvidia-fbc-565.57.01-6.el8.i686
nvidia-modprobe-565.57.01-2.el8.x86_64
nvidia-driver-565.57.01-6.el8.x86_64
nvidia-driver-libs-565.57.01-6.el8.x86_64
kmod-nvidia-565.57.01-3.el8.x86_64
libnvidia-ml-565.57.01-6.el8.x86_64
nvidia-driver-cuda-libs-565.57.01-6.el8.i686
nvidia-libXNVCtrl-565.57.01-2.el8.x86_64
nvidia-settings-565.57.01-2.el8.x86_64
nvidia-driver-cuda-libs-565.57.01-6.el8.x86_64
xorg-x11-nvidia-565.57.01-6.el8.x86_64
nvidia-kmod-common-565.57.01-3.el8.noarch

```
This seems to be a side-effect of the switch to the NVidia OSS GPU driver but it's still a pain.. :(
I do have this modprobe.d file in place:

```
[root@palanthas ~]# cat /etc/modprobe.d/nvidia.conf
options nvidia NVreg_OpenRmEnableUnsupportedGpus=1
```

ElCoyote27, 2024-12-08:
Found some info here:
https://negativo17.org/nvidia-driver/#Proprietary_and_open_source_kernel_modules

Unfortunately there is no akmod rpm for RHEL8, only dkms and kmod. :( Do you think that akmods would work on RHEL too?

scaronni, 2024-12-09:
Yes, but there has never been the request for now.If you want I can add them in the repository. There is also `dkms-nvidia` for EL8/9.

scaronni, 2024-12-09:
Maybe for precompiled shall we stick with the proprietary modules and not the open ones until there is a very good reason to do so?

ElCoyote27, 2024-12-09:
for precompiled kmods on RHEL, yes, I think it would be a good idea to stick to proprietary.

scaronni, 2024-12-09:
ok, will add the `akmods-nvidia` packages also for RHEL.

May I ask why `akmods` and not `dkms` though? I just pushed 3.1.3 which fixes a few annoying bugs on rhel regarding uninstallation: https://bodhi.fedoraproject.org/updates/?packages=dkms

scaronni, 2024-12-09:
So, until now:

- https://github.com/negativo17/nvidia-kmod:
  - `epel-8/epel-9`: kABI kernel module packages
  - `fedora-40+`: AKMOD packages
 - https://github.com/negativo17/dkms-nvidia:
   - DKMS packages

As of now:

- https://github.com/negativo17/nvidia-kmod: AKMOD packages
- https://github.com/negativo17/kmod-nvidia: kABI packages (`epel-8/epel-9` only, proprietary modules)
- https://github.com/negativo17/dkms-nvidia: DKMS packages

All of the 3 variations (kABI, DKMS, AKMOD) provide `nvidia-kmod` and can be exchanged at will as the kernel provider type.

scaronni, 2024-12-09:
Will all come online shortly.

closed 2024-12-09 by scaronni

ElCoyote27, 2024-12-11:
please also note than RHEL-EUS's minor is usually different from RHEL-current about 6 months per year (they match the rest of the year). I hope it's a bit clearer.
There are some nice graphics which show what happens visually at this URL: https://access.redhat.com/support/policy/updates/errata

ElCoyote27, 2024-12-11:
@scaronni I've now transitioned to akmods on all of my RHEL8 machines and it's working very well, THANK YOU! :)

```
[root@daltigoth ~]# rpm -qa dkms-nvidia kmod-nvidia

[root@daltigoth ~]# rpm -qa akmod\* kmod-n\*
akmod-nvidia-565.77-1.el8.x86_64
akmods-0.5.6-24.el8.noarch
kmod-nvidia-4.18.0-553.el8_10-565.77-1.el8.x86_64

```