negativo17/nvidia-driver #169: Multimedia and nvidia repos out of sync

p5, 2025-01-28:
The https://negativo17.org/repos/multimedia/fedora-41/x86_64/ repos are providing packages for NVIDIA 570 whereas the https://negativo17.org/repos/nvidia/fedora-41/x86_64/ is still providing NVIDIA 565.  This causes a mismatch of driver versions on systems and can cause the system to crash.

We've been experiencing this a lot in Universal Blue (https://github.com/ublue-os), where our recent builds have a mixture of versions and users are forced to rollback to the images from a few days ago.

Are these expected to not be in sync?

p5, 2025-01-28:
I've realised there's an akmod-nvidia in both the multimedia and nvidia repos, so perhaps we simply need to force our builds to use the nvidia repo.

p5, 2025-01-29:
This has now been resolved

closed 2025-01-29 by p5

scaronni, 2025-01-29:
Hi @p5 i prepared the builds a few hours in advance of the public release, but open release i forgot to trigger the sync of the `nvidia` repository and did only the `multimedia` one.

Having said that, the newest packages should win, but I guess that's not the case with immutable images.