negativo17/nvidia-driver #170: NVIDIA drivers update on fedora-multimedia dependency problem

jvillanueva-sqone, 2025-02-02:
On the fedora-multimedia repo there's a new version of NVIDIA packages (570.86.16-1), but there's a dependency problem with nvidia-kmod package not found:

Problem 1: cannot install the best update candidate for package nvidia-kmod-common-3:570.86.15-1.fc41.noarch
  - nothing provides nvidia-kmod = 3:570.86.16 needed by nvidia-kmod-common-3:570.86.16-1.fc41.noarch from fedora-multimedia

Seems the problem comes as there's only a kmod-nvidia-570.86.15-1.fc41.x86_64.rpm, not the corresponding 570.86.16-1 version.

On the fedora-nvidia repo the version is 570.86.15-1, so the drivers installation works fine, as all the rpm packages versions on the repo matches.

Vash63, 2025-02-02:
I'm seeing the same. I think for some reason the multimedia repo got an update that the nvidia one didn't. I've been waiting for the nv one to refresh

jp7677, 2025-02-02:
The multimedia repo (https://negativo17.org/repos/multimedia/fedora-41/x86_64/?C=M;O=D) is also incomplete at the moment which prevents updating (or even installing) from there. `kmod-nvidia` and `dkms-nvidia` are still on version `570.86.15` whereas the other packages are updated to `570.86.16`.

zeroepoch, 2025-02-02:
I created a PR for this, https://github.com/negativo17/nvidia-kmod/pull/14, but I won't be able to update the negativo17.org repos. I'm building the package locally to workaround this.

zeroepoch, 2025-02-02:
Here's some instructions for building the `nvidia-kmod` package locally using Fedora's mock tool:
```
$ git clone https://github.com/negativo17/nvidia-driver.git
$ git clone https://github.com/zeroepoch/nvidia-kmod.git  # to pick up the PR changes
$ cd nvidia-kmod/
$ ../nvidia-driver/nvidia-generate-tarballs.sh
$ mock --spec nvidia-kmod.spec --sources $PWD
$ sudo dnf install /var/lib/mock/fedora-41-x86_64/result/*kmod-nvidia-*.rpm
```

jvillanueva-sqone, 2025-02-03:
> Here's some instructions for building the `nvidia-kmod` package locally using Fedora's mock tool:
> 
> ```
> $ git clone https://github.com/negativo17/nvidia-driver.git
> $ git clone https://github.com/zeroepoch/nvidia-kmod.git  # to pick up the PR changes
> $ cd nvidia-kmod/
> $ ../nvidia-driver/nvidia-generate-tarballs.sh
> $ mock --spec nvidia-kmod.spec --sources $PWD
> $ sudo dnf install /var/lib/mock/fedora-41-x86_64/result/*kmod-nvidia-*.rpm
> ```
> 
> If there are other packages that depended on the driver update happening first, then you can try `sudo dnf upgrade` to pick up the rest.

That worked perfectly. Thanks!

scaronni, 2025-02-03:
Sorry got some family issues over the weekend and left the uploads in an inconsistent state as I had to rush away. New uploads coming in a few minutes.

closed 2025-02-03 by scaronni